procount/pinn #647: Share SSH keys between OS's

NumberOneGit, 2022-04-06:
What would be the preferred method of sharing SSH keys between PINN, RPOS and other installed OS's?  Should I create an authorized_keys file in the /settings/dropbear folder?

procount, 2022-04-07:
Nearly. See https://github.com/procount/pinn/blob/master/README_PINN.md#secure-remote-access

NumberOneGit, 2022-04-11:
Thanks!  I'm not sure how I missed that.  

I wanted to use the same private key across multiple OS's and used these steps:

Generate ECDSA key pair and duplicate before conversion.
Convert ECDSA private key to PEM format:  "ssh-keygen -p -m pem -f ssh_host_ecdsa_key" 
Then to dropbear: "dropbearconvert openssh dropbear ssh_host_ecdsa_key dropbear_ecdsa_host_key".   

The PEM formatted key is only used as an intermediary in the conversion process then discarded.  I'm using the original generated ECDSA key for my other OS's.

closed 2022-04-16 by NumberOneGit