procount/pinn #650: Arch2 (Arch Linux) is using removed links

drone540, 2022-04-19:
Hey it appears that the Arch Linux for Raspberry Pi 3 (and anything else referring to Arch2) is still referencing ARMv6 install links and cannot be installed by PINN as ARMv6 is no longer a supported platform.

Please update the link to:
http://os.archlinuxarm.org/os/ArchLinuxARM-rpi-armv7-latest.tar.gz

techpixel, 2022-04-24:
Took me a while, but I found where it was calling the old link: 
https://github.com/procount/pinn-os/blob/c63b133b5f5ff17d5e030ee44b7a304d2e58fe4d/os/Arch2/partitions.json#L19

Try installing it manually using the provided os metadata.

techpixel, 2022-04-24:
If you are using a Raspberry Pi 4B, you can try the Arch4 install using ARMv8: https://sourceforge.net/projects/pinn/files/os/Arch4/

procount, 2022-04-28:
I have removed the ARM6 version and updated ARCH2 for all ARM7 models.
ARM8 (64-bit) uses UBoot, so it will require more work, and may not be suitable for conversion.

closed 2022-04-28 by procount

techpixel, 2022-04-29:
Experiencing issues. Unable to get `boot_arch.tar.xz`.

procount, 2022-04-29:
sourceforge should be working properly again now.