procount/pinn #653: Problem Starting LibreElec in Raspberry Pi 4 with both HDMI connected

Hiforever, 2022-04-25:
LibreElect won't start on either HDMI if there are 2 active HDMI connections on my Raspaberry Pi 4 8GB.   It seems that most of othe time my RP4 stayed in recovery mode.  Occasionally,  LibreElec may start after waiting for 10 or more minutes.   PINN will start LibreElec without problem if there is only one HDMI connection, either HDMI-0 OR HDMI-1, on my RP4.

LibreElect starts without any problem if it is installed as a standalone OS by itself in SD card without installing PINN. 

My configuration:
 
Raspaberry Pi 4 8GB
- HDMI-0 connect to a Marantz AV Amplifier 
- HDMI-1 connect to a LG TV
PINN  3.8.3
LibreElec  10.0.2

procount, 2022-04-28:
I just tried this on my Pi4-4GB, PINN v3.8.3 with LibreELEC 10.0.2. When I booted LibreELEC, the LibreELEC logo appeared on both screens, but KODI appeared on HDMI-0 leaving HDMI-1 black. On Exit, the LibreELEC logo appeared on both screens again.

procount, 2022-04-28:
I tried again with a Pi4-8GB and it still worked the same.

Hiforever, 2022-04-28:
Thanks for checking.

1. What kind of screens are you using ?  In my case, both HDMI are connected to 4k devices (HDMI-0 -Marantz AV Amplifier, HDMI-1 - LG TV).  So if LibreElec logo appears, the resouluion should be 4096 x 2160.
2.  Is there anyway that I can check whether my RP4 is running PINN or LibreElec when it stuck to trace the problem?

procount, 2022-04-29:
1. Hdmi-0 is an 8" HDMIPI lcd screen. 1280x800. Hdmi-1 is 1920x1080 but uses a hdmi to vga adaptor.
2. Do you only have LibreELEC installed with PINN? In which case after PINN's flashing message about pressing the shift key, pinn will reboot into LibreELEC so the subsequnt displays will be from LibreELEC.

Hiforever, 2022-04-29:
1.  The only difference between my setup with your's is the screen resolution.  Both of HDMIs of my RP4 are connected to 4k devices (HDMI-0 LG 4K TV, HDMI-1 AV amplifier connect to the same LG 4K TV in another port) 
 
I will try connecting one HDMI port to a non-4k display later today to see whether this is the cause
   
2. Yes, my LibreElec is installed with PINN.  This was installed with LibreElec as the only OS under PINN.   Nothing output to either display once PINN tried to reboot into libreElec as if the system was hang with occasionally flashing of green light of my RP4.  So LibreElec never get installed if both HDMI are connected.

If LibreElec installed & setup successfuly with PINN if there is only one HDMI connected.  If both HDMIs are connected afterwards, no output to either display once PINN tried to start LibreElec.

As you start LibreElec without any problem with both HDMI connected to non-4k display, it may be an issue that relates to the display resolution.

procount, 2022-04-29:
LibreELEC is provided in NOOBS/PINN format by its maintainers, so I presume it is identical in content to an image file. The only difference is where it is located on the SD card for multibooting.
After PINN selects an OS to boot, it just tells the firmware which partition to reload and reboots, so PINN's presence shouldn't make any difference.
If you add a file called `autoboot.txt` to PINN's partition, with the single line `boot_partition=6`, then the RPi will ignore PINN altogether and boot directly into LibreELEC, thus avoiding any PINN interaction. You will need to remove this file to restore PINN functionality.

Hiforever, 2022-04-29:
thx for your help.  Will try and revert

Hiforever, 2022-04-30:
I've tested various configuration as advise with the following results

LibreElec boots normally without any problem in anyone of the following configurations:

1.   Include autoboot.txt in PINN's partition with the single line "boot_partition=6".   Both HDMIs of my RP4 were connected to 4k devices.

2.  Normal PINN functionalities without autoboot.txt.    HDMI-1 is connected to non-4k display (resolution 2560 x 1080).   HDMI-0 can be connected to a 4k TV or non-4k display.

LibreElec cannot start if both HDMIs are connected to 4k devices.

procount, 2022-04-30:
This is a surprising result. It looks like something PINN is doing is being remembered across a warm reboot. 
To narrow this down, could you please try replacing PINN with previous versions of v3.5 and v3.3, and also with the latest NOOBS? You just need to unzip the alternative versions' zip file over the PINN partition, but please preserve recovery.cmdline from your current version. If you don't, you'll find it wipes out LibreELEC and you would have to reinstall it. In particular please make sure recovery.cmdline does NOT include the 'runinstaller' option.

Hiforever, 2022-04-30:
Can you tell me how to do it ?  

I was able to locate PINN-35.zip and PINN-33.zip from sourceforge archive. However, I don't know how to keep the latest version of NOOBS.

procount, 2022-04-30:
https://downloads.raspberrypi.org/NOOBS_lite/images/NOOBS_lite-2022-01-28/NOOBS_lite_v3_8.zip

To be absolutely correct:
1) Make a copy of recovery.cmdline
2) delete all files on the PINN recovery partition.
3) Unzip the old PINN/NOOBS zip file onto the pinn recovery partition.
4) restore your saved copy of recovery.cmdline
5) Check recovery.cmdline does NOT include 'runinstaller'
6) Boot your card.
7) If PINN detects a new version is available, just say NO when asked if you want to upgrade.
Finally, when all tests are complete, restore the latest PINN v3.8.3, remembering also to restore your recovery.cmdline.

Hiforever, 2022-04-30:
LibreElec starts normally without any problem with both PINN-355.zip and NOOBS_lite_v3.8.zip.  Both HDMIs of my RP4 is connected to 4k display

procount, 2022-04-30:
I can't think of anything after 3.5.5 in pinn that would cause this. Maybe a firmware update...?
If you have time to test the later versions of pinn to determine which is the last version of pinn that works ok, it might help to narrow down the issue.

Hiforever, 2022-05-01:
I couldn't find a version of PINN that works ok with LibreElec in RP4 with both HMDIs connected to 4k devices. 

The earliest PINN version tested was 3.51.   PINN cannot start for versions before 3.51.   For NOOBS, the earliest version tested was 3.3.

procount, 2022-05-03:
Pi4 should work with any pinn version from v3.3 onwards. Earlier versions may not allow downloading of some information due to certificate expiry issues, but that is not relevant here. 
I had advocated restoring the recovery.cmdline, but that was primarily to ensure the runinstaller option wasn't selected. I can't remember if there were any other changes to that file which were required so you could try using the one extracted from each respective zip file, but remembering to delete the runinstaller option.

Hiforever, 2022-05-04:
Thanks.    Will test it in the next few days and revert

lurch, 2022-05-07:
> I've tested various configuration as advise with the following results
> 
> LibreElec boots normally without any problem in anyone of the following configurations:
> 
> 1. Include autoboot.txt in PINN's partition with the single line "boot_partition=6".   Both HDMIs of my RP4 were connected to 4k devices.
> 2. Normal PINN functionalities without autoboot.txt.    HDMI-1 is connected to non-4k display (resolution 2560 x 1080).   HDMI-0 can be connected to a 4k TV or non-4k display.
> 
> Without autoboot.txt included, LibreElec cannot start if both HDMIs are connected to 4k devices.

Very strange. ping @XECDesign and @popcornmix

Hiforever, 2022-05-08:
Tested PINN ver 3.3.4.4 with no luck.  LibreElec cannot start as other PINN versions if both HDMIs were connected to 4k devices.

PINN 3.3 up to PINN 3.3.2 cannot start in RPi4 8GB.     There was a list of error messages saying that "This borad requires newer software"... "recovery.elf is not compatible"   etc

Hiforever, 2023-03-17:
The issue resolve itself after LibreElect version 10.0.4 update.

closed 2023-03-17 by Hiforever