procount/pinn #656: Raspberry pi 3B+

myxxmikeyxx, 2022-05-06:
When I try to put a wap_supplicant.conf file in the root it boots up but then it says no network interfaces found.
Do you have a sample wpa_supplicant.conf?
Is it suppose to have `country=US
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1` at the top?

procount, 2022-05-06:
PINN will copy a default wpa_supplicant file to /settings. You can copy it to PINN's recovery partition as follows:
From PINN's recovery menu, Press `Ctrl-Alt-F2`. Login with username: `root` and password `raspberry`.
Then enter the following commands:
```
root@recovery:~# mount -o remount,rw /mnt
root@recovery:~# cp /settings/wpa_supplicant.conf /mnt
root@recovery:~# sync
```
Now you can turn off the power, remove the SD card and edit the wpa_supplicant.conf file.

procount, 2022-05-09:
> It did not mention how to actually have it in use when you first boot it.

Sorry I don't understand this question. On first boot, any wpa_supplicant.conf file on the root partition is copied to /settings and used for the wifi straight away.

> How to you get back to the GUI after doing Ctrl-Alt-F2

Press `Ctrl-Alt_F1`

> Also if I put a wpa_supplicant.conf file in the root of the SD card and boot then it says no interfaces found and the wifi scanning screen is empty.

This suggests you do not have a driver for the wifi. As you are using the 3B+ it should have onboard wifi and the appropriate driver should be included, although I can't test this myself as my 3B+ has broken. Or are you using a separate wifi dongle?

myxxmikeyxx, 2022-05-09:
So the Wi-Fi on the board works fine when I do a fresh image and manually connect it.
When I add the wpa_supplicant.conf file is when I get these boxes:
Box 1: ` Network access is required to use PINN without local images. Please select your Wi-Fi network in the next scan.`
Box 2: `No Wi-Fi interfaces available` 
Box 3: Wi-Fi Network selection opens up but I can not scan or see any networks.

When I do a fresh image I get the Box 1 and Box 3 and they scan and show available connections.

> This suggests you do not have a driver for the Wi-Fi. As you are using the 3B+ it should have onboard Wi-Fi and the appropriate driver should be included, although I can't test this myself as my 3B+ has broken. Or are you using a separate Wi-Fi dongle?

I am not using a separate dongle. Just trying to use the built in Wi-Fi.

> If you drop into the recovery shell (using ctrl-alt-f2), what is the output of ifconfig?
> Do you get entries for eth0, lo and wlan0?

Yes I see a wlan0 and eth0

`eth0          Link encap:Ethernet            HWaddr  B8:27:EB:AA:7F:B7
                  UP BROADCAST   MULTICAST     MTU:1500   Metric:1
                  RX packets:0        errors:0       dropped:0    overruns:0   frame:0
                  TX packets:0        errors:0       dropped:0    overruns:0   carrier:0
                  collisions: 0         txqueuelen:  1000
                  RX bytes:0 (0.0 B) TX bytes:0 (0.0 B)

lo               Link encap:Local           Loopback
                  inet addr:127.0.0.1  Mask:255.0.0.0
                  UP LOOPBACK RUNNING  MTU:65536   Metric:1
                  RX packets:0        errors:0       dropped:0    overruns:0   frame:0
                  TX packets:0        errors:0       dropped:0    overruns:0   carrier:0
                  collisions: 0         txqueuelen:  1000
                  RX bytes:0 (0.0 B) TX bytes:0 (0.0 B)

wlan0         Link encap:Ethernet            HWaddr  B8:27:EB:FF:2A:E2
                  UP BROADCAST   MULTICAST     MTU:1500   Metric:1
                  RX packets:0        errors:0       dropped:0    overruns:0   frame:0
                  TX packets:0        errors:0       dropped:0    overruns:0   carrier:0
                  collisions: 0         txqueuelen:  1000
                  RX bytes:0 (0.0 B) TX bytes:0 (0.0 B)
                  `

lurch, 2022-05-09:
Have you got your Wifi country-code correctly set in `wpa_supplicant.conf`? https://www.raspberrypi.com/documentation/computers/configuration.html#configuring-networking-2

procount, 2022-05-09:
>So the Wi-Fi on the board works fine when I do a fresh image and manually connect it.
When I add the wpa_supplicant.conf file is when I get these boxes:

If it works fine doing it manually, why are you bothering to create another wpa_supplicant.conf file? 
Why not set it up manually, then drop into the recovery shell and execute the following:

```
mount -o remount,rw /mnt
cp /settings/wpa_supplicant.conf /mnt
sync
```
You will now have a copy of your working wpa_supplicant.conf file in the recovery partition that you can copy and reuse elsewhere.

>Also I can not get Ctrl + Alt + F1 to go back to the GUI.
Can I run a command to reboot it because reboot doesn't work and exit just gets me stuck.

If `ctrl-Alt-F2` and `ctrl-alt-f3` work, I've no idea why `ctrl-alt-f1` would not work. You are using it locally and not via VNC, aren't you? You can use `rebootp` to reboot out if necessary.

> Have you got your Wifi country-code correctly set in wpa_supplicant.conf?

Actually, the country code does not seem necessary in PINN (Unlike in Raspberry Pi OS). So try removing that, especially as you said it works manually. But what channel is your wifi router using?

procount, 2022-05-09:
> When I add the wpa_supplicant.conf file is when I get these boxes:
Box 1: Network access is required to use PINN without local images. Please select your Wi-Fi network in the next scan.

You may still get Box1 if your wifi is working but PINN takes a long time to get a DHCP lease from your router. But in this case you should see the list of OSes being populated behind and an IP address showing in the title bar. You should increase the network timeout settings to avoid this error if that happens.
I get Box2 on my 3B where the wifi is broken.

myxxmikeyxx, 2022-05-09:
I do not know what it was but I recreated the file in VSCode and put it on the SD card and now it works and is connected. 
Sorry for all the trouble.

Although when I try installing playerpipper it is breaking and crashing the install.

closed 2022-05-09 by myxxmikeyxx

lurch, 2022-05-09:
Ahhh, maybe the file needs Unix line-endings and you were creating it with Windows line-endings?

procount, 2022-05-09:
Glad you sorted it. Maybe it was the Windows/Linux line ending differences? 🤷‍♂️ 

Playpiper continued on #655

myxxmikeyxx, 2022-05-09:
I think that it was something like that. I saw somewhere mention to create it using notepad++ or vscode and not regular notepad. I don't know why but it worked. 🤷‍♂️