procount/pinn #657: Convert os backup to bootable image

techpixel, 2022-05-06:
Is it possible?

lurch, 2022-05-07:
By "bootable image" I assume you mean a `disk.img` file that you could write to an SD-card with e.g. RPi Imager.

Short answer: No, it's not possible because NOOBS/PINN work in a different way to standalone images, see https://github.com/raspberrypi/noobs/wiki/Standalone-partitioning-explained and https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained for more info.

Long answer: If you **really** know what you're doing, then it's theoretically possible - you'd basically need to do all of https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN but in reverse. And if you do want to go down this path, you may also find bits of https://github.com/RPi-Distro/pi-gen useful (which is the software used to build the images on https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-32-bit ).

techpixel, 2022-05-07:
Thanks for the information.

closed 2022-05-07 by techpixel