procount/pinn #660: Issue accessing JSON file

eroom1966, 2022-05-11:
@procount 
Hi, I have had successful usage of pinn since 2019, last issue I had was here
https://github.com/procount/pinn/issues/343
all of a sudden this week it broke for me 
I have changed nothing at my end, and so I am completely confused, but now when trying to re-install I get the following message

[image]
If I use the same URL in a curl/wget command on another machine, it works fine

not sure how to debug this, the path reported is

https://pilotaware.lode.co.uk/downloads/OGN-Beta/net/os_list.json

I can send a link to the installer zip file if that would help to reproduce

Any help greatly appreciated
Thx
Lee

jose1711, 2022-05-11:
Just guessing.. HTTPS uses certificates so I wonder if the clock shows valid date, can you check?

eroom1966, 2022-05-11:
Ahh, OK good question 
- how do I do that ?

Is this done at the server end or client ?
BTW, I noticed that all the references in the JSON files, and the cmdline were referencing http not https, but I guess the message in the dialog, was after a redirect from http to https

if you can let me have the commands to check the above, I can paste them here
Thx
Lee

eroom1966, 2022-05-11:
@jose1711 good catch!

I tried this in the 'online browser' in the installer, and it reported
certificate expired

it also asked if I wanted to continue anyway.
Is there any way to ignore certificate errors in the installer ?

thx
Lee

eroom1966, 2022-05-11:
It appears to be an issue with the installer - If I test the access to the website outside of the RPI/installer, I do not get an issue. But the installer browser reports a problem

Is this a date/time comparison question ?
where is the installer getting the current date and time to perform this comparison as it does not have a realtime clock

Also I checked the certificate by going here
https://www.digicert.com/

and entering the website address
pilotaware.lode.co.uk

all seems fine at the server end
Thx
Lee

eroom1966, 2022-05-11:
This version of the installer I am using on the remote machines was built as follows

```
Build-date: 2019-11-21
PINN Version: v3.3.2
PINN Git HEAD @ f88d00f107202418e73409b048315839fbd743c0
rpi-userland Git master @ de4a7f2e3c391e2d3bc76af31864270e7802d9ac
rpi-firmware Git master @ 64b5649a41b69d09bfe0ed05448d28a66be3edfd
rpi-linux Git rpi-4.19.y @ 71d47f4c4bd7fd395b87c474498187b2f9be8751
```

Is it possible there is an issue with that release, that will not now allow me to perform the remote recovery ?

eroom1966, 2022-05-11:
OK, I think I have resolved the issue.
In the version of pinn I currently have deployed in the field (2019-11-21), this can no longer access the repos, producing an SSL cert error, even though there is no SSL cert error 
I just tried integrating my changes into pinn version 2022-03-17, and now it can access my repos again, which has the same SSL certificate

so for others who have an earlier version of pinn in the field that they cannot push these changes - they may no longer work

jose1711, 2022-05-11:
Sorry for late answer - I was not at the desk. I downloaded PINN v3.3.2 and was able to reproduce the problem. Next thing I did was starting a recovery shell - there I invoked: `wget https://pilotaware.lode.co.uk` only to get an error message about expired Let's Encrypt certificate. This is a client-side issue (certificate is valid as you mentioned already and date/time was synced fine).

In order to fix this client's (PINN's) certificate store must be updated with the current list of valid CAs. Not sure if PINN has a mechanism to make that happen. Also the documentation fails to mention this could happen so it's probably better to keep this open until the documentation (at least) gets an update.

lurch, 2022-05-11:
Yeah, one of the LetsEncrypt root certificates expired a while ago, see https://github.com/procount/pinn/search?q=letsencrypt&type=issues for more details.

eroom1966, 2022-05-11:
reopened as requested

procount, 2022-05-11:
This problem was documented in github issues and on the forum thread. It was a bit of a surprise and caught many embedded devices out.

It was fixed a while ago, so I don't see any need for any further documentation.