procount/pinn #662: RUNINSTALLER - initialize drive (blocking)

eroom1966, 2022-05-11:
Hi All

I have installed a new version of pinn, and I am trying to create an unattended install, which is currently blocking with a popup saying 

RUNINSTALLER - Initialise Drive
There are existing OSes installed Are you sure you want to initialise the crive
Yes | No

The answer is YES, but as it is unattended - I need to automate the response, otherwise the install will hang awaiting input
how do I overcome this ?

Searching issue list for other answers ....
Thx
Lee

eroom1966, 2022-05-11:
I see an option forceruninstaller, trying this now, I presume this is an option to runinstaller, rather than an alternate command ?

eroom1966, 2022-05-11:
answering my own question - this seems to fix the unattended installation

closed 2022-05-11 by eroom1966

lurch, 2022-05-11:
Forgive me for saying, but it sounds like you're trying to do things with PINN that it wasn't really designed to do?

eroom1966, 2022-05-11:
@lurch @procount 

I think this was originally one of the intentions, here is a discussion from 2019 where I discussed this use model with the developer
https://github.com/procount/pinn/issues/343

I have it working again having updated the version, and used the additional command line option

FTAOD - if you have remote devices in the field, which you cannot attend to install a new/modified OS, then this is a very good method of maintenance

thx
Lee