procount/pinn #663: Error building PINN 3.8.3 (missing binary operator before token)

jose1711, 2022-05-11:
```console
./BUILDME.sh
..
  CC       fd-hook.o
  CC       fd-safer-flag.o
In file included from /usr/include/signal.h:328,
                 from ./signal.h:52,
                 from c-stack.c:49:
c-stack.c:55:26: error: missing binary operator before token "("
   55 | #elif HAVE_LIBSIGSEGV && SIGSTKSZ < 16384
      |                          ^~~~~~~~
  CC       dup-safer-flag.o
make[4]: *** [Makefile:1915: c-stack.o] Error 1
make[4]: *** Waiting for unfinished jobs....
```
OS: Arch Linux x86_64 rolling
gcc version: 12.1.0

lurch, 2022-05-11:
You might need to try building PINN on an OS with an "older" version of GCC?
(I don't think PINN will be fixing this "bug" itself, because it relies on a lot of code from the upstream [buildroot](https://buildroot.org/) project)

procount, 2022-05-11:
PINN's development environment has only recently changed from Ubuntu 14.04 to Ubuntu 20.04 to match NOOBS. Such a change is always a challenge, so if you want to build on anything else then I'm afraid you're on your own.

closed 2022-05-11 by procount

jose1711, 2022-05-11:
I tried with `gcc` 10, 8, 7 - all stopped on the same error. By the looks it seems coming from this bit in `/usr/include/bits/sigstksz.h`:
```
 26 /* Default stack size for a signal handler: sysconf (SC_SIGSTKSZ).  */
 27 # undef SIGSTKSZ
 28 # define SIGSTKSZ sysconf (_SC_SIGSTKSZ)
```
The file is part of `glibc` 2.35 on my system. If I temporarily change line 28 to
```
# define SIGSTKSZ 8192
```
then the error vanishes. So it  seems to be related to version of `glibc` rather than the compiler.

jose1711, 2022-05-11:
FWIW I got quite far with the above modification and setting `gcc` in `Makefile`:
```
HOSTCC := gcc-10
HOSTCXX := g++-10
```

The build failed in later phases when `libiconv` was about to be installed:
```
libtool: link: /../pinn-p3.8.3/buildroot/output/host/usr/bin/ccache /../pinn-p3.8.3/buildroot/output/host/usr/bin/arm-buildroot-linux-uclibcgnueabi-gcc -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Os iconv.o -o iconv  ../srclib/libicrt.a /usr/lib/libiconv.so
/usr/lib/libiconv.so: file not recognized: File format not recognized
collect2: error: ld returned 1 exit status
```
Perhaps this helps someone - I am stopping here. Prebuilt images work for me and I can use vagrant of Ubuntu if I need more. Thanks.

procount, 2022-05-11:
FWIW, I think Ubuntu 20.04 uses gcc 9.4.0