procount/pinn #668: Could you please add Ubuntu Desktop 22.04 LTS?

name01019, 2022-06-10:
Ubuntu has been updated to 22.04
https://ubuntu.com/download/raspberry-pi

paulwratt, 2022-09-29:
not sure if this helps some

http://cdimage.ubuntu.com/ubuntu-server/jammy/daily-preinstalled/current/

AlarmingBox, 2022-10-31:
yes please add it. was getting some errors upgrading from the previous version. Apparently 22.04 is the first version of Ubuntu Desktop certified for Raspberry Pi 4. 
https://ubuntu.com/blog/whats-new-for-raspberry-pi-ubuntu-22-04-lts

lurch, 2022-11-03:
ping @waveform80 to see if there's any interest at Canonical for providing Ubuntu in NOOBS/PINN-format tarballs?

procount, 2022-11-04:
OK it's coming! Basic conversion is done but needs some tidy up.

procount, 2022-11-06:
Ubuntu 22.04.1 Desktop for Pi4 is now available for installation.

closed 2022-11-06 by procount

miguelgargallo, 2023-07-05:
Yes please, still 2023 without server option on PINN menu, Ubuntu 22.04.2 LTS (Jammy Jellyfish) server

procount, 2023-07-07:
Ubuntu 22.04.2 Desktop & Server versions now added.

procount, 2023-07-09:
EndeavourOS does not have an RPI image and creating it is quite lengthy, even after a PINN install.
Could you install Arch and then install the desktop on top?

procount, 2023-07-17:
Great!
I'm not having much luck creating a PINN version.
Would a version without desktop be sufficient?