procount/pinn #669: Install os from USB Drive

XZCendence, 2022-06-15:
How on earth do you install an os .img.xz file from a USB on to the SD card with pinn? I have mine in the /os folder but pinn is not recognizing them. My OS images do not have separate tarballs for root and boot partitions, it's just all one big img.xz file. What do I need to do to have pinn actually recognize them, I've been trying to get this to work for 5 days, it should not be that hard.

procount, 2022-06-15:
You don't. PINN does not install images. It installs tarballs. Blame NOOBS! 😉 
Why?
In order to multiboot OSes PINN needs to store the OSes on different partitions. Image files contain partition information AND the file contents. If you installed image files they would all be fighting to be stored on the same partitions. So PINN creates new partitions and just untars the tarballs onto them.

So first of all you need to convert your image file by separating the partition and file contents and creating tarballs out of the file contents. Then add the additional meta-files to tell PINN all about your new OS.

Please see https://github.com/procount/pinn/blob/master/README_PINN.md and https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN. You can also look at the existing installations on sourceforge for examples of the required files.

The meta-files need to include a partition-setup.sh script that configures your OS for its new partition layout. Which OS are you trying to install? Each OS can have different requirements. If it uses U-Boot, that must also be removed.

XZCendence, 2022-06-17:
I'm using the pwnagotchi re4son image, found [here](https://github.com/evilsocket/pwnagotchi/releases), I'm trying to dualboot it with kali 64bit. I see that PINN already has kali, so I just need to convert my pwnagotchi image to PINN format.