procount/pinn #670: Cant Press Backup Button

Retro-Devils, 2022-06-17:
So ive made a Retropie os and a Android 11 os. Installed through PINN and everything . Now i wanna backup but i cant press the button, its just grey.

Retro-Devils, 2022-06-17:
update . i formated a2 tb drive using pinn and made a os folder on it still cant press backup. Ive also now noticed i cant download. 

So i went in recovery with crt+alt+f2 and seen it didnt mount my drive fomatted with pinn and same results didnt mount my drive

Retro-Devils, 2022-06-17:
So i got a backup error when i finally was able to backup. Any help?

procount, 2022-06-17:
What was the backup error? (See troubleshooting on the wiki)

Retro-Devils, 2022-06-17:
ill pull error from log now

Retro-Devils, 2022-06-17:
says "error writing ......... Disk Full? 

it a 2 tb usb images are on a 256 gb card.

procount, 2022-06-17:
FAT32 is limited to 4GB files. Maybe try formatting it as EXT instead.

Retro-Devils, 2022-06-17:
ok ok . ill try now

Retro-Devils, 2022-06-17:
worked great!!

closed 2022-06-20 by Retro-Devils