procount/pinn #672: Installation impossible

Rommell761, 2022-06-29:
### Discussed in https://github.com/procount/pinn/discussions/671

<div type='discussions-op-text'>

<sup>Originally posted by **Rommell761** June 29, 2022</sup>
Hi , I am trying since yesterday to make a fresh dual boot installation ( batocera / lineageos 19  and raspbian / lineage os 19 )
But i continuosly getting this error. 

Anybody have idea how to solve it ? 

[image]

Thanx in advance for any help 
</div>

Rommell761, 2022-06-30:
> It's probably a sourceforge mirror issue. It may resolve itself in a day
> or so.
> Which of those OSes are causing the problem?

Hi , thanks for your anser, It seem with lineage os 19

procount, 2022-06-30:
It appears Sourceforge has chosen the telkom university in Indonesia to download LineageOS for you, Probably because it is the nearest mirror to you. I have downloaded a few of the Lineage19 files from there without error, but there could still be a problem with that mirror.
You could try manually downloading all the files and folders from the sourceforge folder at https://sourceforge.net/projects/pinn/files/os/lineage19-rpi4/, copy then to a USB stick and install them from there.
If you still have trouble downloading them, you can click on the "problems downloading?" button and choose a different Mirror.
Please refer to https://github.com/procount/pinn/blob/master/README_PINN.md#local-images-on-usb-stick on how to store them on a USB stick and install the OS from there.