procount/pinn #676: ssh to pinn boot

The-Exterminator, 2022-08-01:
How can i ssh to the boot menu from my PC. ??? so i dont need USB keybord and mouse

procount, 2022-08-01:
Ssh us for a text interface but the boot menu is graphical so you must use vnc. I suggest to add vncshare and forcetriggee. See https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-use-pinn-headlessly-vnc

The-Exterminator, 2022-08-01:
> Ssh us for a text interface but the boot menu is graphical so you must use vnc. I suggest to add vncshare and forcetriggee. See https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-use-pinn-headlessly-vnc

i can now ssh to the pinn, but how can i start the pinn menu, so i can choise it from my pc. ???

procount, 2022-08-02:
Like I said above, you can't from ssh. You must use VNC.
Download a vnc client such as tightvnc or realvnc for your PC and use that instead of ssh.

closed 2022-08-04 by The-Exterminator