procount/pinn #678: Double-clicking checkboxes does weird things

jojomoore2007, 2022-08-05:
Applies to: Raspberry Pi 4, PINN Lite
# Steps to Reproduce

1. Download PINN Lite
2. Extract it to a microSD card
3. Put the microSD card into the Raspberry Pi
4. Boot the Raspberry Pi
5. Connect to a WiFi network
6. Wait for options to load
7. Double-click any item's checkbox
8. One of the following things happens:
    1. A different box is checked, and the clicked box is unchecked.
    2. No boxes are checked, but the needed space is non-zero.
    3. The clicked box is checked, but the space necessary is doubled, as if the box was selected twice.
    4. Two boxes are checked.

I happened to stumble on situation 2, but have had no luck reproducing it yet. If I manage to do so in the near future, I will post the steps here.

win8linux, 2022-08-08:
I can replicate the following on Raspberry Pi 1B+ and 3B:

*   No boxes are checked, but the needed space is non-zero.
*   The clicked box is checked, but the space necessary is doubled, as if the box was selected twice.
*   Two boxes are checked.

procount, 2022-08-08:
I haven't been able to reproduce any of this.
Are you using a mouse, touchscreen or joystick/gamepad to double-click? which make/model?
If no boxes are checked, the Needed space should be zero. Are you sure you haven't any other OSes ticked in another category? What happens if you click on the clear button?