procount/pinn #679: Distros not downloading and/or installing

christyler80, 2022-08-14:
First, thank you for all of the work this has gone into Pinn OS.

Unfortunately, I have encountered the following problems when trying to use PINN on my 8GB Pi. I have encountered exactly the same problems using both of my 240GB SSDs and which work fine with other Linux distros.

1) the PINN Lite list of available OSs is not consistent - e.g. sometimes Kali and Manjaro are listed - sometimes they are not listed.

2) after installing OSs using PIN Lite, when then booting Raspberry OS for the first time, the Raspberry OS splash screen appears briefly, before the screen goes blank and the SSDs light starts to blink on/off. This goes on for hours.

3) Using the web installer https://pinn.mjh.nz/, although I can select Kali and Manjaro along with (say) Rasperry OS 64bit, Twister and Ubuntu Mate 64bit, Kali and Manjaro were not downloaded. The other 3 distros did download, but I then ran into issue 2).

Any help would be appreciated.

Many Thanks

Chris

procount, 2022-08-14:
1. The list of OSes is downloaded from the internet from various repositories, so it is subject to the vagueries of the internet and its servers. If there are errors or timeouts on any of the servers, then the OSes it hosts may not be presented or installed.
2. The Raspberry Pi OS version installed by PINN is created by the RPF - it is the same one you would get using NOOBS, or installing the image directly. Once PINN has installed it, it reboots directly into Raspberry Pi OS, taking no part in the boot process itself, so it is unlikely that PINN is the cause of this. I have only experienced it once recently myself directly after updating the eeprom firmware and installing the latest Pi OS. Have you seen this with other OSes? My feeling is it could be something to do with the video and the reading of the EDID codes from the monitor. This has changed a lot recently in PI OS and some monitors have issues with the new way video modes are selected.
3. The "Web" installer on Matt's page caches the list of available OSes, so all OSes should be listed. However that doesn't mean their servers are all operational, but you will only find this out when you try to download the OSes. Many OSes are stored on sourceforge which does do maintenance occasionally, during which time there can be errors. Also some of their mirrors can have issues.
4. Be careful with the PI4 8GB. Some OSes may predate that model and not have the required firmware version to run, In this case, replacing the firmware from the latest Raspberry Pi OS and maybe adding additional drivers should sort it.
5. I'm afraid I don't have the same time (or hard disk space)available to keep on top of updating all the OSes all the time, so some get a bit out of date. If there are any particular ones you have issues with, please try their image version directly to ensure that the OS will work normally, then let me know and I can try and update it.

christyler80, 2022-08-15:
Hi,

 

Many thanks for your comprehensive reply and I will try again over the next few days.

 

Many Thanks

 

Chris

christyler80, 2022-08-15:
> Hi,
>
>
>
> Progress, in that I have Raspberry OS, Twister and Ubuntu Mate all installed and working fine. Forcing 1080p using config.txt on each distro helped, as I can see the installation progress now!!
>
>
>
> Kali 64bit and Manjaro (xfce) are not downloading when I select them from Matt’s page, nor do they appear in the PINN application ‘install’ list. That is a bit odd, as I was able to download both Kali 64bit and Manjaro (xfce) at one point yesterday, but not now – although there are on the SourceForge <https://sourceforge.net/projects/pinn/files/os/>  page.
>
>
>
> Anyway, I note your ‘README’ files on GitHub and I will have a go, at downloading the Kali 64bit <https://sourceforge.net/projects/pinn/files/os/kali20.04-64/>  and Manjaro (xfce) <https://sourceforge.net/projects/pinn/files/os/manjaro-xfce-rpi4/>  files from SourceForge <https://sourceforge.net/projects/pinn/files/os/>  manually.
>
>
>
> That said, any ‘quick start’ advice on how to do this and/or a way of ‘forcing’ the download of Kali 64bit <https://sourceforge.net/projects/pinn/files/os/kali20.04-64/>  and Manjaro (xfce) <https://sourceforge.net/projects/pinn/files/os/manjaro-xfce-rpi4/>  from within the PINN application – e.g. by modifying a .json file, would be appreciated.

Hi,

 

If I copy the contents of pinn-lite to a SD card and boot from that, then Kali and Manjaro are available and I can install them to my USB SSD.

 

However, if I unplug the SD card, copy the contents of pinn-lite to my USB SSD and boot from that, Kali and Manjaro are not available.

 

Thanks

 

Chris

procount, 2022-08-15:
Probably because they dont support booting from usb.

christyler80, 2022-08-15:
I already have both Kali and Manjaro booting from USB SSD drives on my Pi.

procount, 2022-08-15:
They have probably upgraded those OSes to support usb/ssd since I converted them. I will put them om my list to upgrade them, but it may be a little while.

christyler80, 2022-08-15:
I will complete my installation tonight and confirm (hopefully) that all is working tomorrow. 

 

Thanks, Chris

christyler80, 2022-08-16:
I cannot get Manjaro to boot from USB, although the Debain based distros Kali, Twister, Raspbian and Ubuntu were OK.

 

Really, I am looking for a solution that will allow any Pi compatible disto to run via a multi boot manager, in much the same way as (say) Ventoy <https://www.ventoy.net/en/index.html>  on Windows.

 

I appreciate though, this is far easier said than done.

procount, 2022-08-16:
Largely it is dependent on the OS/Distro,.PINN is just the means to install it, configure it and select it. The booting of the OS is done by the RPi firmware itself.
The OS must be capable of booting from a different partition to work in a multiboot scenario. To boot from USB/ssd, it has to be able to reference its partitions by partuuid or other means, instead of assuming all OSes are booted from /dev/mmcblk0p1. Not all can.
And if it uses Uboot, that needs to be stripped out, unless you can use Uboot as your boot manager.
Pinn does not deal with GPT disks or UEFI based OSes.

christyler80, 2022-10-11:
Thank you and I will let you know if I get anywhere….

strolch101, 2023-01-05:
I can confirm @christyler80 experience with the missing availability of Manjaro KDE when PINN is on USB and OS installing target also is USB. In contrast, if PINN is on SD-Card and installing target is USB, Manjaro shows up, but my multi-os installation failed to boot afterwards. However, Manjaro ARM itself is pretty much able to boot from USB stand-alone on Raspberry Pi 4. Tested it right now with an SD-Card Image(!) written onto a USB-Drive and booted the Pi with only the USB-Drive attached.