procount/pinn #684: Use on Raspberry Pi with external BT/2.4GHz keyboard

lavinkabul, 2022-09-01:
Use on Raspberry Pi with external BT /2.4GHz keyboard

I have a Raspberry Pi 4B 4 GB, I want to install a PINN multiloader to select the boot of two systems, Batocera and Android TV. It is possible to "play around" windows 10. I will install everything on a 64GB SD card from a PC (Raspberry Pi Imager). The raspberry will be connected to a Samsung 1920 x 1080 TV.
I want to buy a D8 Plus wireless keyboard ( https://aliexpress.com/item/1005001638954955.html?_ga=2.208392850.1239333740.1661638378-835044333.1660590148&sku_id=12000016930735858&gatewayAdapt=glo2rus ), there is the possibility of working with a 2.4 GHz adapter.
So the question is, will this keyboard work "out of the box" without additional dancing with a tambourine, such as searching for drivers? In theory, all such keyboards are unified, but I would like to know for sure.

procount, 2022-09-01:
PINN does NOT support Bluetooth, so no Bluetooth accessories (keyboard mouse etc) can be used with it.

The link suggests this keyboard additionally has a 2.4GHz wireless dongle. In the past I have used a 2.4GHz wireless mouse successfully. In this case the dongle required no drivers as it appears to the kernel as a USB HID device. So if this is the case with this keyboard as well, then it should work, but I cannot guarantee it.

mickhaorex, 2022-09-01:
Thank you, I understand you. I will order this keyboard and when it arrives by mail, in about two months, I will write about the results here.

closed 2023-03-25 by procount