procount/pinn #685: Pinn 3.6.2 won't update

jzbond65, 2022-09-06:
I'm stuck on Pinn 3.6.2  the 3.8..3 update downloads just fine but after reboot still old version. Any sugestion how to fix this?

procount, 2022-09-06:
https://forums.raspberrypi.com/viewtopic.php?p=1922284#p1922284

realOgenki, 2022-11-05:
I have the same problem, but I think the PINN partition is "SETTINGS" or not.
The mnt directory corresponds to a boot partition of one of my operating systems.
Should I copy upgrade.sh file in this partition or in another?
Thank you.

procount, 2022-11-05:
By the PINN partition, I mean the partition where PINN is stored which is labelled as the RECOVERY partition. It is the first partition, normally identified as /dev/mmcblk0p1.
If you put your SD card in a Windows machine it will probably be the only partition that is shown (although latest Windows versions might also show the additional BOOT partitions.) The Recovery partition can also be recognized by having the file recovery.rfs on it. This is where you should put upgrade.sh.
After copying upgrade.sh to the RECOVERY partition, you should put your SD card back into your Pi and boot into PINN, pressing SHIFT to make sure you enter the recovery menu. Then you can enter the recovery shell by pressing ctrl-alt-F2 and follow the rest of the instructions given above.

realOgenki, 2022-11-06:
I sorry but I havent got any RECOVERY partition on my PINN SSD card.
I'm using Linux too on my destop:
[image]
And this is my root directory connecting PINN over SSH:
[image]
What do I doing wrong?
Thank you.

procount, 2022-11-06:
You're doing nothing wrong, except you didn't mention you were using Linux on your desktop! :wink:
"RECOVERY" is handled as a special partition name and is normally hidden on a LINUX PC, which is why you cannot see it.
You will need to do this differently. Let me create some different instructions for you.

realOgenki, 2022-11-06:
My apologies, RECOVERY partition isn't mounted by default on my Linux desktop and I wasn´t able to see it. 
Now I have updated my PINN.
Then, is the meaning of SETTINGS partition the boot partition of PINN? 
Thanks

procount, 2022-11-06:
> Now I have updated my PINN.

Do you mean you have been able to successfully run the upgrade.sh script? 
Does that mean you do not need any further instructions?

> Then, is the meaning of SETTINGS partition the boot partition of PINN?

No. As I said above, the RECOVERY partition, being the first partition on the drive (/dev/mmcblk0p1 or maybe /dev/sdb1 on your linux desktop.....?) is PINN's boot partition.
The SETTINGS partition contains settings for PINN about what is installed and some installation meta files - stuff that needs to be stored without affecting the RECOVERY partition, as originally the RECOVERY partition was read-only .

realOgenki, 2022-11-07:
Yes I could. Thanks.
My next challenge will be to reinstall one of PINN's operating systems.
If I can I would like to make a suggestion: show version of OSes to install.

procount, 2022-11-07:
Have you tried hovering your mouse over the OS?
It will show a little yellow popup tooltip with the release date and version (if it was provided)

realOgenki, 2022-11-10:
Thank you. I  hadn't realized the yellow label until now... but I can see "New versión" text next to any of my OSes of PINN.
Excuse me, where is it? I have some OSes upgradables.

procount, 2022-11-10:
If you want to UPGRADE an OS, I suggest you use the upgrade procedure that you normally use for your OS, e.g. apt update/apt upgrade/apt full-upgrade/apt dist-upgrade or whatever your distro recommends.
For major versions it is not always easy or recommended to upgrade in this fashion and then you have to start all over again by installing the latest version and then reinstalling all your apps and data from your backup.

PINN only shows the latest version of each OS (unless they are individually versioned), So if PINN indicates there is a new version of an OS, then you can reinstall the latest version from the maintenance menu. 
HOWEVER, THIS IS DESTRUCTIVE! IT WILL DELETE ALL YOUR APPS AND DATA FOR THAT OS AND REINSTALL A FACTORY FRESH VERSION OF YOUR OS. -use with caution.

realOgenki, 2022-11-12:
Yes, I am aware of the risks when using the reinstall option. First I will make a backup of my operating system data. Thank you for your advice and for this wonderful tool.

closed 2023-03-25 by procount