procount/pinn #686: Issues with ubuntu 22.03

CallMeAryan, 2022-09-08:
i managed to install ubuntu 22.03 by in-built updater in ubuntu 20, it boots up fine with a HDMI screen however i have a raspberry 7" screen. It has same issues like linage os. Display turnes bright white and then fades away but it fixes when we turn on touchscreen mode in linagw settings.  It lowers resolution of OS,  i tries that by connecting it with HDMI but 640x480 isnt avaliable. Im assuming this is because of higher resolution than 480p

Is there any way to resolution override or fix it you used older ubuntu versions to make it work?

Thank you

procount, 2022-09-08:
Sorry, I have no idea. This is more a question you need to take up with Ubuntu.

CallMeAryan, 2022-09-09:
UPDATE:-
I played with setting a bit and saw some other configs, switching to fkms v3d graphic driver made it work. However if it wont work with hdmi display unless you switch to kms again

You can disable kms by commenting out amd adding

#dtoverlay=vc4-kms-v3d
display_auto_detect=1
dtoverlay=vc4-fkms-v3d

and switch to kms again by removing comment

dtoverlay=vc4-kms-v3d
#display_auto_detect=1
#dtoverlay=vc4-fkms-v3d

Ubuntu 22.03 is now perfectly working fine with pinn for me

Thank you 
Have a great day

closed 2022-09-09 by CallMeAryan