procount/pinn #691: No signal from PINN on waveshare 10.1 screen

MazenS, 2022-09-29:
I've got it plugged into my rpi400, when I use rpi os with the recommended config settings:

hdmi_group=2 hdmi_mode=87 hdmi_cvt 1280 800 60 6 0 0 0 hdmi_drive=1

Everything is fine. I reflashed the sd card to put PINN to dual boot and I'm not seeing any output on the monitor.

Ive updated the config for PINN to include those lines but kinda lost now.

Any help would be brilliant.

MazenS, 2022-09-29:
I've also tried the config suggestion here but no luck.

https://github.com/procount/pinn/issues/367

procount, 2022-09-29:
It's not clear to me whether its the PINN display you are not seeing on your waveshare, or the display of an OS installed by PINN.

Please could you post a copy of your config.txt file from your PINN partition and enclose it in triple backticks so it displays nicely.

Remember that each OS (including PINN) is self contained, so you need to apply the display setting to each one seprately.

MazenS, 2022-09-29:
Hello!
Apologies for the lack of clarity.
When booting into the pinn menu from a freshly installed sd via the rpi image program, nothing is being displayed on my above mentioned screen.

When I switch out the sd card to my rpi os install everything outputs fine.

I had edited the initial pinn config and added
`hdmi_group=2 hdmi_mode=87 hdmi_cvt 1280 800 60 6 0 0 0 hdmi_drive=1`
Which were the magic lines to make the screen work with standard rpi os. Raspian. Not sure of its name these days.

procount, 2022-09-29:
Where and how exactly you put those lines makes a difference, which is why I suggested you post the whole config.txt file.

I assume PINN displays ok for you on another HDMI device?

MazenS, 2022-09-30:
I was on my phone last night - not the most conducive to well thought out posts.

`gpu_mem=16
start_file=recovery.elf
fixup_file=fixup_rc.dat

[pi4]
start_file=recover4.elf
fixup_file=fixup4rc.dat
max_framebuffers=2

#hdmi_group=2
#hdmi_mode=87
#hdmi_cvt 1280 800 60 6 0 0 0
#hdmi_drive=1

hdmi_group=2
hdmi_mode=28
hdmi_ignore_edid=0xa5000080`

PINN is displaying ok when I go through my PC monitor.

Thanks for taking the time to help - it is appreciated.

procount, 2022-10-01:
Is it this device -> https://www.waveshare.com/wiki/10.1inch_HDMI_LCD#Software_Setting
That page suggests it is 1024x600.

MazenS, 2022-10-01:
This is the device:
https://www.waveshare.com/10.1inch-hdmi-lcd-b-with-case.htm

And I found the display instructions here: https://www.waveshare.com/wiki/10.1inch_HDMI_LCD_(B)_(with_case)

procount, 2022-10-01:
Well, I'm at a bit of a loss. 
As it uses a HDMI connection it should just work. It's probably just a case of setting it up in config.txt.
I would suggest enabling ssh, vncshare and forcetrigger in PINN and connecting to it via putty over ssh. Then you can try experimenting with tvservice  and fbset etc.

MazenS, 2022-10-01:
Ok I'll take a look after lunch.
I'm a bit of a noob - maybe not a bit - could you explain a little more what I should experiment with?

MazenS, 2022-10-01:
So after some fun and games learning about SSH I have the following
From tvservice:
```state 0x6 [DVI DMT (87) RGB full 4:3 x2], 640x480 @ 59.00Hz, progressive```

From fbset:

```
mode "640x480"
    geometry 640 480 640 480 16
    timings 0 0 0 0 0 0 0
    rgba 5/11,6/5,5/0,0/16
endmode
```

paulwratt, 2022-10-04:
Its been a few years since I played with it, but as far as I remember, there is still a quirk (side effect) of setting the HDMI mode at boot (in config.txt), and that is that resulting queries to the VC for mode and monitor resolutions is _not_ that same as when you let the (HDMI?) monitor choose its _prefered_ resolution, and then run the same queries.

MazenS, 2022-10-04:
I commented out the relevant lines from PINN.
When I ssh back in I'm seeing similar reports:

```
root@recovery:~# tvservice -s
state 0xa [HDMI DMT (4) RGB full 4:3], 640x480 @ 60.00Hz, progressive
root@recovery:~# fbset

mode "640x480"
    geometry 640 480 640 480 16
    timings 0 0 0 0 0 0 0
    rgba 5/11,6/5,5/0,0/16
endmode

```

Is there anything else you could recommend I could try?
I've got in touch with waveshare customer support too - just incase they've seen this before