procount/pinn #695: Raspberry Pi 4 (4GB) and pinn_lite

SerdarARIKAN, 2022-11-09:
Hi there...

I've been reviewing pinn lite for a few days now. Downloading Operating Systems and installing offline seemed like a more practical solution. However, in all my tests, I encountered problems during or after the download of Arch2, TLXOS, piCorePlayer, XBian_RPi3, batocera_RPi4 operating systems. In my experiments, I saw that the same problems continued even though I deleted the entire os folder where the problems were constant and started over. I ran into some issues while following the link below. There are inconsistencies in the respective json files and the os.json files downloaded into the downloaded OS folders. For example, the problem may occur because the Pi 4 is not included in the supported_models line in the os/Arch2/os.json file. I think there may be problems during the processing of the repo_list.json file in the link below. I've also noticed that some operating systems are categorized incorrectly, albeit trivially. For example, since there is no "group": "Games" line in the file named os.json in the Lakka_RPi4_64 folder, the operating system is placed in the general category. I recommend that you review it.

https://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json

Have a nice day.

procount, 2022-11-09:
Are you downloading these OSes manually yourself? Or have you selected the Showall option within PINN?
Many of the OSes you list are not suitable for the Pi4 and PINN will not normally show them or allow you to install or download them. So it is not surprising you are having problems installing them if you are circumventing their selection. If the Pi4 is not listed in those OSes it is because they are not suitable for the Pi4. 
Arch2 is a rolling distro which is automatically updated by its maintainers and I haven't checked it in a while - it could need updating. 
TLXOS and Lakka_RPi_64 are 2 OSes that were part of NOOBS and controlled by the RPF or their maintainers. So they don't have the additional group categories in their json files. However, I work around this by having an override file that is downloaded to override certain features.
It strikes me that maybe you have never connected PINN to the internet and therefore you have not had the opportunity to have the override file be downloaded. Had you done so, Lakka would show up in the correct category.
I suggest you connect PINN to the internet and let it download the list of OSes. This will also download the override file. Then power off PINN and reboot it a second time to make sure the override file has taken effect.
If you insert a FAT32 formatted USB drive with a folder called 'os' on it, you can then use PINN's download feature to download the OSes you want to the USB drive. You can then install them from the USB drive without need for the internet at your leisure.

SerdarARIKAN, 2022-11-10:
2,127 / 5,000
Translation results
Hello.
No; I'm not downloading it myself and the show option isn't selected either. That's exactly what I'm trying to explain.
Since the beginning of my experiments (https://github.com/procount/pinn/blob/master/README_PINN.md), I have been using the official download method in your documents with the user interface of pinn. I am aware of the overrides.json file and its function, although I still do not fully understand how some of the stages work.

When the error occurred, I tried to follow the process manually starting from the recovery.cmdline file in the pinn_lite.zip file, starting from the address http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json, and I tried to trace some errors from the related json files. and i realized it could be due to the processing of these files. I think there is something wrong with your review.

Although I follow the official documents, even if there is a problem beyond my knowledge and control, all of the operating systems that you said should not be listed on RPi4B by pinn_lite, except TLXOS and batocera_RPi4, continue to be listed for RPi4B on the online pinn at https://pinn.mjh.nz/ . However, I don't think this is a major issue. Because as far as I understand (please correct me if I'm wrong) for example at http://downloads.raspberrypi.org/os_list_v3.json TLX seems to be supported by RPi4B but actually it shouldn't. Or at https://raw.githubusercontent.com/matthuisman/pinn-os/master/os_list_v3.json it seems that batocera_RPi4 is supported by RPi4B, but it shouldn't actually be. It is very easy to delete them from relevant places.

The main problem is why the error occurs on other operating systems.

I can't do a detailed review right now but I will report any errors for you by doing a fresh download using the official method. Then I believe we can make a more detailed diagnosis.

If you have time, I recommend you to download all operating systems for RPi4B via the user interface, and then just check whether all operating systems are downloaded in the interface and whether the downloaded operating systems are in the correct categories, without even trying to install.

procount, 2022-11-12:
I did spot some issues with the categories when following your use case of downloading to USB and then reloading from there without the internet which have totally confused me. I'll have to take a closer look.

SerdarARIKAN, 2022-11-13:
Yes, there are problems with categories, but the problem does not end there.  I've also spotted some issues with repos that have been updated independently of you.  Json files may need to be thoroughly reviewed.

procount, 2022-11-16:
I finally solved the mystery of why Lakka_RPI4_64 changes categories when downloaded to USB. 
It's because the meta-data for this OS is different in the online file (os_list_v3.json) than the offline file (os.json). I think the updates in os.json were missed.
These files are on the RPF server, so I will either have to try and get them modified or provide an override for them.

I will check the other OSes you have mentioned, but since I now offer over a hundred different variants of OSes, I won't have time to check all of the rest, but if you spot any more with issues I will take a look at them. Otherwise they will have to wait until they are refreshed with a later version.

SerdarARIKAN, 2022-11-19:
Hello there.
I am getting the following errors while downloading.
In my examination, I detected these files from their links in the os_list.json file. The manual downloads I made worked fine. I guess. An error occurs while parsing the relevant links or wget cannot process the links correctly.

[image]

```
Error downloading or extracting tarball
wget: unable to resolve host adress     'boot_arch.tar.xz' (Arch2)
                                        'archlinuxarm-rpi-2-latest.tar.gz' (Arch2)
                                        'xbianboot-rpi3.tar.xz' (XBian_RPi3)
                                        'xbianroot-rpi3.tar.xz' (XBian_RPi3)
                                        'boot.tar.xz' (batocera_RPi4)
```

Your recovery.cmdline file;
`runinstaller quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline repo_list=http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json loglevel=2 sdhci.debug_quirks2=4`

Your repo list
`http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json`

Arch2 repo
`https://raw.githubusercontent.com/procount/pinn-os/master/os/os_all.json`

os_list.json Arch2 part
```
{
            "os_name":	    "Arch2",
            "description":  "An Arch Linux port for ARM7 devices RPi2/3/4/z2",
            "release_date": "2018-01-28",
            "feature_level": 0,
            "supported_hex_revisions": "1040,1041,2082",
            ### **"supported_models": [**
                "Pi Zero 2",
                "Pi 2",
                "Pi 3",
                **### "Pi 4"**
            ],
            "supports_backup": "update",
            "url":      "https://archlinuxarm.org/platforms/armv6/raspberry-pi",
            "group":    "Minimal",
            "download_size":    466759248,
            "os_info":          "http://downloads.sourceforge.net/project/pinn/os/Arch2/os.json",
            "partitions_info":  "http://downloads.sourceforge.net/project/pinn/os/Arch2/partitions.json",
            "icon":             "http://downloads.sourceforge.net/project/pinn/os/Arch2/Arch2.png",
            "marketing_info":   "http://downloads.sourceforge.net/project/pinn/os/Arch2/marketing.tar",
            "partition_setup":  "http://downloads.sourceforge.net/project/pinn/os/Arch2/partition_setup.sh",
            "tarballs": [
                                "http://downloads.sourceforge.net/project/pinn/os/Arch2/boot_arch.tar.xz",
                                "http://archlinuxarm.org/os/ArchLinuxARM-rpi-armv7-latest.tar.gz"
            ],
            "nominal_size": 1330
}
```

tarball links
```
http://downloads.sourceforge.net/project/pinn/os/Arch2/boot_arch.tar.xz
http://archlinuxarm.org/os/ArchLinuxARM-rpi-armv7-latest.tar.gz
```

and also the os.json file and os_list.json files in your repo links are not consistent. The os_list.json file produced with the help of the recovery.cmdline file is different, for example, the Arch2/os.json file in your sourceforge project.

and finally the sourceforge download counts of the files can give you an idea of which downloads are having problems.

[image]

As you can see, the boot_arch.tar.xz file was downloaded only 57 times, but the partition_setup.sh file was downloaded 102 times. Because an error occurs while downloading the boot_arch.tar.xz file. Other repos have similar issues.

The list of operating systems that I have detected repo-related problems during the download and installation phase is as follows;

```
Arch2
batocera_RPi4
debian64pi4
Lakka_RPi4_64
OSMC_Pi4
piCorePlayer
Recalbox-_-Pi4
recalbox7-rpi4
Retropie4 - dsi
TLXOS
ubuntu22041_64
Xbian_RPi3
```

have a nice day...

closed 2023-11-01 by SerdarARIKAN