procount/pinn #696: Wrong kano os version

keboid, 2022-11-20:
I prepared pinn for a raspberry pi2b, also checked for kano os. As it turned out kano for rpi3 got installed so it wont launch. A rpi2 version of the os also exists.

procount, 2022-11-20:
Kano was prepared by Matt and should be the latest version. It may not work on the original Pi2, but it should work on the latest Pi2 v2 that has the same 64 bit processor as the PI3.
The version for the original Pi2 is from 5/3/2019 and uses Raspbian Jessie, so it is quite old.
Are you still interested in this version? If so I could have a go at converting it.

keboid, 2022-11-20:
If its not a big trouble. I installed also ubuntu mate and full rasbian, cant wait to play around with them, i only used is basically for openelec/libreelec for now. My daughter might love kano.