procount/pinn #698: Ubuntu > 20.10 gives black screen

knachte, 2022-12-03:
I've tried installing a version of ubuntu with pinn 3.8.3 on a pi4 8gb version, booting from an ssd over usb (agron one m2) , next to raspberry pi os and recallbox (which both work), but anything above 20.10 does not start up, seemingly not even giving a hdmi signal. I've tried upgrading from 20.10 to a higher version, but due to it being EOL that is a problem too. 
I tried switching to the console, but that is not showing anything either. I also tried on the other hdmi output, but nothing there either. Any suggestions what i could try?

procount, 2022-12-05:
I have just tried installing Ubuntu_22041_64 on my Pi400. No issue.
I then put this SD card in a USB adapter to simulate an SSD and booted it from USB on a Pi4 8GB. Also no issue.

Could this be just a video issue?
What resolution does your monitor use?
Can you use a different monitor?
Could you try switching to fkms? For this you need to find config.txt on the Ubuntu boot partition and replace dtloverlay=vc4-kms-v3d with dtoverlay-vc4-**f**kms-v3d

knachte, 2022-12-05:
It was hooked up to an LG 27 inch full hd monitor from a couple of years ago (27MP37vq-b). I've now hooked it up to my main 4k screen and and another 1080 screen and there it seems to work ok so it seems to be a combination of ubuntu and that specific screen.
Closing this issue as it clearly is not related to Pinn.

closed 2022-12-05 by knachte