procount/pinn #700: PINN w/ Lineage OS ATV (KonstaKANG) 19.0 for RPi4 -- Problem Installing MindtheGApps12.1 . Alternatives?

arlo-infinity, 2023-01-03:
Greetings!  Please forgive me if this isn't the correct place to bring up this issue.

A few months ago I created a multi-boot image with PINN including lineage19-rpi4-atv amongst several other OSes I was trying out.
At the time I noticed that when following the recommendations in KonstaKANG's release notes I was unable to flash the recommended GApps distro, MindTheGapps-12.1.0-arm64-ATV-20220912_160414.zip , after booting into TWRP.   It failed with a unable to mount "/mnt/system" error which I discovered also happens with some other devices but none of the suggested solutions worked or seemed viable after many hours/days of trying to find a solution.   As I had many other OSes to try out,  and I had previously created a standalone SD card image with lineage19-rpi4-atv which I was able to _successfully_ install the same MindTheGapps distro using the TWRP procedure and this resulted in a working distro, I decided to temporarily just swap SD cards when I needed my Android TV install vs. some other OS.

I've now tried out all my OSes and I'm returning to this problem.   I can confirm that using the lateset versin of PINN, with a install config created from PINN System Resize I still have the same issue.  I can boot into Android and use settings to boot into TWRP, but the setup fails to mount "/mnt/system" when trying to install MindTheGapps-12.1.0-arm64-ATV-20220912_160414.zip .   Open GApps isn't available for Lineage 19/Android 12 and so far the one distro I was able to install "NikGApps" results in a black screen with a cursor after booting and does not lead to the expected "New Android TV Device Setup" process which occurs when the same files are installed standalone outside of PINN.   

I am guessing that the usual culprit of the MindTheGapps install script assuming a basic Lineage partition setup which isn't the case, but I don't have any idea how to either patch the MindTheGapps script or otherwise find another workaround to install the default script.  Is anyone aware of a workaround that allows MindTheGapps to be installed into lineage19-rpi4-atv on PINN. ?

After using PINN for some time now I am also aware that it is usually possible to create custom "xxxxx.tar.gz" partition images and install them over existing distributions, for example, when you have customized your OS significantly and don't want to have to do everything over again.  I have read the "Installing 2 custom OSes into PINN" instruction thread. I've successfully done a hack version of this myself for BATOCERA where I used the default PINN installer to make new install, including a pristine BATOCERA install, and then extracted an archival backup of each relevant boot.tar.gz / root.tar.gz / data.tar.gz to the corresponding partition to restore the customizations to the new install.   Once I figured out how to tar up a partition preserving ownership and permissions this proved relatively straightforward.   I think maybe I edited the os_config.json file to match the partition layout but I don't actually know if that was critical.   Is it possible to "backup" and "restore" the partitions of my standalone SD card with Lineage OS 19 Rpi4 ATV on it in a similar way?  This would bypass the need to install GApps on a fresh PINN install of lineage19-rpi4-atv.

I am aware that since it has 4 partitions and for other reasons Lineage is more difficult to deal with within PINN than a typical OS.    I found the original thread from 2017 between KonstaKang and procount which indicated that "SELinux contexts" must be preserved when creating images using .tar.gz archives, as well as procount indicating they found a workaround that allowed them to port Linegage into PINN in 2017, but no specifics were mentioned.    I know that that the file /vendor/etc/fstab.rpi4 needs to be modified to reflect the correct 4 Lineage partitions in use.   Is this is the only place that needs to be updated beyond os_config.json?  I have been looking for other locations on the /boot partition that need updating because I can't figure out how the /vendor/etc/fstab.rpi4 file is actually read since vendor is (relatively) the third consecutive partition if /boot is the (relatively) first partition.   Is there somewhere else on /boot that provides the absolute or relative partition map?

If this isn't the correct place to point out this issue or ask for assistance I would appreciate if you could point me to a better place.  Thanks in any case.

procount, 2023-01-03:
Yes, this is as good a place as any to raise this issue, and thanks for doing so so comprehensively.
It has been a while since I actually tested the installation of GAPPS on Lineage. As the OS conversion had gone ok and it booted under PINN I thought GAPPS would install ok. Konsta had warned that the changes in Android would make it more and more difficult to multiboot Lineage, but it still seemed to be working.
I see Lineage20 (Android 13) is now available so I need to update it anyway. Maybe I should try converting that and testing GAPPS on it, Would that be suitable, or do you need version 19?

arlo-infinity, 2023-01-03:
From my perspective Lineage 20 (Android 13) _would_ be great but as far as I can tell in my searches for different GApps versions the Lineage-developer recommended GApps packages (MindtheGapps or OpenGapps) aren't yet available for Android 13.  As of today KonstaKang has a warning to this effect and doesn't have a link.   I suspect that you can't install the GApps 12L versions on Lineage 20 because I think Android 13 will complain -- but I am happy to check if it works on a standalone SD card if you want.   I see a NikGapps has a version for Android 13 but so far for me I haven't been able to get the NikGapps Android12L version working on Lineage 19 OS (at least the PINN installed version), and so far NikGapps isn't a recommended package by either KonstaKang or the LineageOS team.    

I completely understand you may want to wait for MindtheGapps or OpenGapps for Android13 to be released before you put the time in to do the the Lineage20 conversion, and you may not want to bother with the Lineage19 now that Lineage20 is out.
In that case, is there any hope of me doing the surgery of transferring my exisiting working SD card that has Lineage19+GApps working to the 4 partitions on my multiboot, or are there so many places where the partition mapping has changed after the install that it is too much of a pain?

If you will go ahead with trying to see if you can get MindtheGApps to be installed on PINN-installed Lineage19 OS please let me know if I can do anything to assist.

Thanks for your help.

Veda82, 2023-01-20:
@arlo-infinity I had the same issue. I can confirm that following the procedure you mentioned ("Installing 2 custom OSes into PINN") I could create a multi boot environment.

1) Installed Lineage 19.1 in a separate SD card
2) Installed MindtheGapps latest version
3) creating a custom os following the procedure

Regarding 3, I started from the files in https://sourceforge.net/projects/pinn/files/os/lineage19-rpi4-atv/.

At first I didn't have success, probably because I backed up system and vendor partition tarring the filesystem: then I created an img of the 2 partitions using dd, compressed those img with xz and sobstituted those files in the original Lineage 19 installation package from sourceforge.

I also remove md5 in the partition json and updated the size of the uncompressed file.

Regarding userdata: I did a fresh installation so I used the package from sourceforge (wish I guess is basically an empty partition).

Hope this helps.

Regarding Lineage 20: as you said it would be cool to have it in PINN, but for now without Gapps it makes little sense...

procount, 2023-01-20:
> At first I didn't have success

You have to go through a few hoops to get Lineage to recognise the new partition layout, and Konsta has indicated it may not continue to work in future versions.

arlo-infinity, 2023-01-22:
> @arlo-infinity I had the same issue. I can confirm that following the procedure you mentioned ("Installing 2 custom OSes into PINN") I could create a multi boot environment.
>
> 1. Installed Lineage [...]

Total synchronocity!  

I was just about to post about my similar successes in getting my setup working a couple of days ago (using a different method) and some of the things I learned.

TLDR:  
1. Issue is more generally with the MindTheGApps install script and PINN rather than Lineage19ATV and PINN, as the previous version Lineage18ATV works as expected using TWRP to flash OpenGApps_11.0_xxxxxx.gz but _fails_ with the same error if you install MindTheGApps_11.0_xxxxxx.gz onto Lineage18ATV as when installing MindTheGApps_12.1_xxxxxx.gz onto Lineage19.  
2. I was able to transfer my customized (with GApps and various other apps installed) Lineage19ATV SD card over to a Multiboot system freeware DiskGenius on Windows and its "Clone Partition to Image File" and "Restore Partition Image File" features -- "Clone Partition" also works.  I cloned all of "boot", "/", "vendor", and "userdata", restored them to the multiboot, then edited "os_config.json" on boot partition and "/etc/fstab.rpi4" on the vendor partition to map the partitions correctly.
3.  After restoring the partitions it didn't work and was boot looping PINN over and over.  Only after I used PINN to "fix" the filesystem (dirty bit set wrong) and rerun partition_setup.sh did it work.  !Warning!:  If you boot back into TWRP for whatever reason it will only boot into TWRP unless you edit other fstab files.  I didn't bother fixing this as I don't need TWRP anymore.

More Details:

It sounds like were able to get a fresh install of Lineage 19.1 working with GApps this way, which is what most people would want.  I _think_  you are saying that at first you didn't have success because you backed up vendor and system by tarring them up, but then had success by making an .img of system and vendor and integrating them with the sourceforge distro, and using that custom distro when doing a PINN install.  Did I understand correctly?  

I never got as far as going through and making a nice custom distro that plays nicely with the PINN installer as I was just too impatient.   What I did do originally was to try and tar up all 4 partitions one at a time from my standalone SD card installation (that included MindtheGApps) and was created from the KonstaKang image not the procount image from sourceforge --- and then restore them to a multiboot installation that included Lineage19ATV  This led to a bootloop where PINN kept attempting to boot Lineage19ATV and I gave up at that point.  Giving up may have been a mistake, I may have been closer than I thought, as I discuss below.

Since I couldn't get my custom Lineage19ATV working previously I decided to backtrack and try a multiboot using Lineage18ATV instead, as I was primarily focussed on the Android TV functionallity.   Doing a OS "replace" operation on an existing multiboot I was able to follow the default instructions from KonstaKang's page and use TWRP to flash OpenGAPPS_11.0_xxxxxxx_tvstock.xxxxx.gz no problem and setup Lineage18ATV (Android TV 11 ) super easily.   BTW:  As an experiment I tried to flash MindTheGapps onto Lineage18ATV and it threw up the same error as in Lineage19ATV, so the problem is the script for MindTheGapps in general, NOT Lineage19ATV necessarily.  It is very possible that if there were an OpenGAPPS_12.1_xxxxxxxx_tvstock.xxxxx.gz it would just work fine with Lineage19ATV, and a  similar statement holds for OpenGAPPS_13.0_xxxxxxxx_tvstock.xxxxx.gz and Lineage20ATV.

Once I had Lineage18ATV working on a multiboot customized it a little and decided to experiment with transfering this version (that I knew worked on multiboot) to a different multiboot.    This is when I decided to try DiskGenius as I remembered seeing the options to backup and restore individual partitions to image files.   It seems to backup the partition at the INODE level.
I cloned all 4 partitions from on SD card to a different multiboot, adjusted the "os_config.json" on the boot partition and "/etc/fstab.rpi4" on the vendor partition and to my surprise IT STILL DIDNT WORK.  As a last resort, which maybe is obvious to some, I tried using the "fix" menu option to fix the file system and re-run partition_setup.sh and after running that it booted no problem.   These leads me to suspect that it might have worked fine with the tar/untar each partition method if I had just used the "fix" menu there too.

So now encouraged by this new discovery I tried using DiskGenius to transfer my customized (with GApps and various other apps installed) Lineage19ATV SD card over to a Multiboot.   Performing the same procedure of backing up and restoring images of each partition, and editing the  "os_config.json" on the boot partition and "/etc/fstab.rpi4" on the vendor partition, and using "fix" menu to fix the filesystem and re-run partition_setup.sh it WORKED and booted into the Android TV 12L system properly.
As a test I decided to boot into TWRP and this resulted in the system being stuck booting TWRP, however if you have already installed GApps (and whatever software you need TWRP to install) there is not any good reason to boot into TWRP again.
[... cut ...] em and re-run partition_setup.sh 

If you aren't comfortable doing either of those options then Lineage18ATV will work without issues and I'd recommend using that for now.

If for some reason OpenGApps won't be doing full releases for Android 12L then it might be worth trying to fix MindTheGapps script, but if we do expect OpenGApps to be released "soon" then I think it is better to wait and see.     The same goes for Lineage20ATV (Android TV 13) --- not urgent unless we have GApps of some sort.

procount, 2023-01-23:
Some additional info that might help you doing these conversions:
1. System and Vendor partitions are created as image files since tarring them loses some file attributes (to do with selinux I think)
2. Boot and Userdata partitions can be tarred
3. If you look at partition_setup.sh, you'll see the partition references that need updating are buried in the ramdisk files on the boot partition. Running the fixup script after you have installed the new images/tar files should fix this up for you, as you have seen.
Unfortunately, I cannot distribute LineageOS with GAPPS or MindtheGapps and I don't have the time nor experience to fix MindTheGapps installation scripts for PINN.

arlo-infinity, 2023-01-23:
> Some additional info that might help you doing these conversions:
>
> 1. System and Vendor partitions are created as image files since tarring them loses some file attributes (to do with selinux I [...]

1.  I see.   I was looking into the SELinux attribute preservation awhile ago before I tried DiskGenius.   It seems with the current version of tar are the right OS you can use "tar --selinux --acls --xattrs  -cvzf file.tar.gz /filesystem/tobackup" to preserve SELinux, ACLs, or all extended attributes.   
3.  Thanks for the explaner.  That makes way more sense now.

Yes, if people could distribute it with GApps I suppose it would already be done that way.  My only suggestion is to give people a heads up (maybe a link to a note on github that has a pass through link to KonstaKANG's page) if they are installing Lineage19 or Lineage19ATV that it is normal for MindTheGApps not to install, in the case of Lineage19 they should try and use OpenGApps_12.1 "stock" and in the case of Lineage19ATV until OpenGApps "tvstock" for 12.1 is released for now they either have to do something like Veda82 or I have done or just use Lineage18ATV if they want a working Android TV install using KonstaKANG's instructions.

I took a look at the MindTheGApps install script (specifically the file "update-binary" and it looks like it uses some "find partitions automatically" function that is used to parse "/etc/recovery.fstab" when inside TWRP.   This looks like a good candidate for the failure point, but I also don't have the experience to fix it without putting in too much time to figure it out.

Thanks again for PINN and your help!

Veda82, 2023-01-23:
@arlo-infinity:

> I think you are saying that at first you didn't have success because you backed up vendor and system by tarring them up, but then had success by making an .img of system and vendor and integrating [...]

Yes, that's exactly what I did. I (lazily :) ) tried the procedure as it is (i.e. tarring the partition) because I was not sure on how to create the img file, but I found out that dd was enough.

@procount 

> 1. System and Vendor partitions are created as image files since tarring them loses some file attributes (to do with selinux I think)

Yes, I found a previous discussion you had with the creator of Lineageos when selinux was mentioned, and at that point I figured that that could be the reason and I tried imging the partition as it was done in the os repo, and everything was ok.

Veda82, 2023-01-23:
Well, it seems Gapps for Android TV 13 are available:

http://downloads.codefi.re/jdcteam/javelinanddart/gapps/ATV

Haven't tried those yet...

arlo-infinity, 2023-01-24:
> Well, it seems Gapps for Android TV 13 are available:
> 
> http://downloads.codefi.re/jdcteam/javelinanddart/gapps/ATV
> 
> Haven't tried those yet...

@Veda82 
Released today! What timing...   

I just tried installing "MindTheGApps for Android TV 13" on Lineage20ATV and it _didn't_ work for me.   The setup TV launcher never ran and it just boots to a blue screen.  If you get a chance to try it let me know if it works for you.  I'm going to try and see if anyone has brought this up with KonstaKang, and if not I guess I will.   

Also, it seems as though OpenGApps has no plans for releasing new versions currently, so we may be stuck with trying to get MindTheGApps working one way or the other if we want Android 13+ ATV (on PINN or otherwise).

Veda82, 2023-01-24:
Won't be able to try until a couple of days... I'll let you know. You may want to write on the XDA thread/Konsta website.

Veda82, 2023-01-24:
Straight from XDA:

https://forum.xda-developers.com/t/dev-rom-unofficial-lineageos-20-android-13-for-raspberry-pi-4-b.4516545/page-3#post-88043899

Seems there's a workaround...

arlo-infinity, 2023-01-25:
@Veda82 

> Straight from XDA:
> 
> https://forum.xda-developers.com/t/dev-rom-unofficial-lineageos-20-android-13-for-raspberry-pi-4-b.4516545/page-3#post-88043899
> 
> Seems there's a workaround...

FYI: I tried the workaround but it didn't initially work as I and someone else on XDA find we just get a black screen (and in my case at least a bootloop) after patching the "update-binary" script as KonstaKANG instructed.   I then tried a diff of the 12.1 "update-binary" and the new "update-binary" and I found that there were a few other lines of script changed very recently.   I decided to try rolling this changes back as well (actually, I just copied the 12.1 "update-binary" into the 13.0 package, and rezipping it) and this WORKED.  Just posted the news on XDA.

Incidentally, I'm not sure if you are already aware but I just learned that the new January 9, 2023 release of Lineage20ATV from KonstaKANG has H.264 Hardware Video Decoding/Encoding enabled by default for the first time ever in any of his Raspberry Pi builds.  This was my only major complaint about using it as a daily drive for TV.  So far I have seen zero glitches with HD video streaming (which was NOT the case previsouly even with the software workaround from Lineage19.1ATV) and so this is a big deal and a huge imrovement to software decoding -- and it looks like KonstaKANG has a preliminary version working for H.265 decoding as well.   It gives me further motivation to try and get GApps working with Lineage20ATV+PINN somehow by learning how to patch the "update-binary" script.   Of course we should be able to transplant a standalone Lineage20ATV install to PINN as we did for Lineage19ATV --- but it sure would be nice if anyone could just use it seamlessly.   IMHO Lineage is definiately one of the major use cases for having PINN installed.

@procount 
With these very recent developments (especially the hardware video decoding) it looks as though it would be quite useful to take Lineage20ATV to the same point as Lineage19ATV is in the PINN ecosystem.    I know you have said you don't have the time to do it yourself, but I will be looking more closely into the "update-binary" script in the hopes of finding a way to make it play nice with PINN.    If there is any way you could take a short peak at "/META-INF/com/google/android/update-binary" (of  MindTheGapps-12.1.0-arm64-ATV-20220912_160414.zip) through the lens of your expert knowledge of PINN and how it works with partitions it would be super appreciated.   Any insight could be very helpful.    It is reporting an errior mounting "/mnt/system" and the relevant lines are just below line 187.   If you really don't have the time or have no clue what could be happening that's fine too, I just thought I would ask.   Thanks again for making PINN!

Veda82, 2023-01-25:
@arlo-infinity Yes, I was aware of the H264 hardware decoding... In fact it was the main motivation I was looking into LineageOS, particularly as a potential Geforce Now client... Very nice to hear about H265 as well!!!

Thanks for your investigations, I will definitely try it in the next few days!!!

Of course having official support in PINN would be wonderful... Don't know how I could help, I'm not a coder even if I have some experience with Linux scripting... Anyhow, for the time being I can use your workaround.

Thanks again!