procount/pinn #703: Chromium OS

trehousecomicsDev, 2023-01-14:
Would it be possible to use the empty data partition to add chromium os to pinn? If so how can this be accomplished? If it could be added to the installer website that would be even better

procount, 2023-01-14:
I'm pretty sure I tried to convert a version of Chromium before, but it did not lend itself to multi-booting under PINN.

trehousecomicsDev, 2023-01-14:
Its based on Gentoo so I would think it would work

procount, 2023-01-14:
Are you referring to the FlintOS / FydeOS version of Chromium OS?
IIRC, these used U-Boot as a loader and GPT as a partitioning scheme. These need removal or adaption to suit PINN.

trehousecomicsDev, 2023-01-14:
https://github.com/FydeOS/chromium_os-raspberry_pi/releases/tag/r102-r2 this one preferably. This is the stock build by fydeos not the Fydeos with the subsystem

trehousecomicsDev, 2023-01-14:
What kind of partition scheme is required?

trehousecomicsDev, 2023-01-14:
https://www.chromium.org/chromium-os/developer-guide/disk-layout-format/

trehousecomicsDev, 2023-01-14:
[image]

trehousecomicsDev, 2023-01-14:
> Are you referring to the FlintOS / FydeOS version of Chromium OS? IIRC, these used U-Boot as a loader and GPT as a partitioning scheme. These need removal or adaption to suit PINN.

They uses stateful

procount, 2023-01-15:
[image]
Unfortunately, this is not suitable for PINN due to the highlighted areas.

trehousecomicsDev, 2023-01-15:
[image]
How does this work on the pi 3 what would need to happen to make it work on the Pi 4

procount, 2023-01-15:
Pi3 and earlier images tended to be specific for the Raspberry Pi, using the RPi firmware bootloader and an MBR partition scheme. These are the types of images that NOOBS and hence PINN were designed to install.

To minimise the number of different images needed to be produced for an OS, maintainers of ARM64 images preferred to standardise on the images, so they use ubiquitous bootloaders such as U-Boot or UEFI so that the main Kernel and Distro is common. Unfortunately, these don't work well with PINN's multibooting.

At the same time, larger SSDs required GPT format, or sometimes this was just used for ease once the RPi firmware supported it. PINN only works with MBR.

In some cases U-Boot can be removed. I am not familiar enough with UEFI to know how to convert or remove it. It's possible that GPT can be converted to MBR, but this needs investigation on each OS.

The other alternative is to upgrade a PI3 image to be suitable for the PI4. This involves upgrading the RPi firmware, but sometimes requires new kernel drivers to match the new hardware.

If you can find a Chromium OS image without these limitations, I would be happy to try converting it.

closed 2023-03-25 by procount