procount/pinn #709: Compressed Images

symbios24, 2023-03-02:
Hello there

Is possible to add an option to have the images/operating systems compressed to save space???

Like for example in berryboot if you know it that the operating system are compressed with the LZO method.

Thank you

procount, 2023-03-02:
All OS installation files are compressed using the highest level xz compression for quick download and minimal storage. Other compression types are also supported e.g. lzo, gzip, zip, bz2 etc. OS backups are compressed using gzip due to resource issues.

symbios24, 2023-03-02:
I mean once installed in the card to be compressed,they are uncompressed in installation right??

symbios24, 2023-03-02:
For example raspbian desktop image in berryboot is only 1GB  in the card and raspbian lite in 350mb

symbios24, 2023-03-02:
Is it possible to take out the berryboot project anyway,is hasn't been updated for over 1.5 year from what i know its creator has abandoned it
Its a great project it will be great if someone with your experience can manage it

Thanks

procount, 2023-03-03:
Berryboot is a great program and was the precursor to NOOBS/PINN, but it works quite differently as it keeps the OS as squashfs files and uses overlays to store the changes, so they all share the same partition. PINN allocates separate partitions to each OS.

symbios24, 2023-03-03:
I tried to build it but it took hours and hours never finishing and most of the time was stopping with errors ussually missing depends so i gave up coz it did the same over and over, raspberry pi zero 2 W does not work with it,also the kernel for Rpi4 is quite old so it does not work correct with newer operating systems.

Its a shame its a very good software 

Thanks

closed 2023-03-25 by procount