procount/pinn #710: RPi 4B PINN not starting on USB SSD

ajl199, 2023-03-02:
Thanks procount for advice over on the rapberrypi forums. I'm battling forward with a Raspberry Pi 4B and 1TB SSD in USB enclosure. However, I can't get PINN to start correctly from USB. I see in the quick start [documentation](https://github.com/procount/pinn/blob/master/README_PINN.md#quick-start) that boot and root fs on USB is "only applicable to RPi models based on the BCM2837 chipset (RPi3B and RPi2Bv1.2)", but there seem to be lots of people booting 4Bs from USB, including with PINN, in contradiction of that so I've been trying and this one does boot, just gets stuck subsequently. Does the documentation post-date the 4 meaning I do actually need to use an SD card to boot from on the 4B? If so, question ends here.

If not, and USB boot on the 4B should work, the situation is that if I write PINN to an SD card, the Pi boots into it and goes through to network selection etc. If I put it on the USB SSD, it sometimes boots to the PINN backdrop with no window of any kind shown, sometimes does that and then displays the error "Error saving boot files to memory. SD card may be damaged.", sometimes shows a "Saving boot files to memory" progress bar for a bit and then either freezes or displays the previously mentioned error, and sometimes throws "Cannot find the drive with PINN files". In none of these cases does the main PINN window with "Install", "Wifi", "Clear" etc. show. The SSD has an MSDOS filesystem and single fat32 partition containing PINN (same files as on the SD card) on it, created through gparted. I've dosfsck'd it several times and the filesystem seems to be alright. I've tried making the partition the full TB and only 1GB, but that makes no difference; there seems to be no rhyme or reason about which errors come up when.

I tried putting the working  SD card in a USB adapter as a means of debugging, and that failed at "USB MSD timed out..." before even reaching the PINN GUI.

Am I being entirely dim here and doing something plain wrong, or does this sound like hardware trouble? If the latter and anyone wants to put bets on which bit of hardware is likely to blame (Pi, SSD or SSD enclosure), or suggest possible workarounds, I'd welcome all ideas.

closed 2023-03-02 by ajl199

procount, 2023-03-02:
Ahh, that would explain a lot!
Many issues on the forums are often solved with a better power supply :)

procount, 2023-03-03:
I've converted OpenPlotter32 for you. Next will be OpenPlotter64 then BBN, Hopefully have it done in a few days, provided I can find enough free time.

ajl199, 2023-03-03:
Thank you! I really appreciate that.

procount, 2023-03-04:
OpenPlotter & BBN are now available. You will find the in the new Sailing tab.

ajl199, 2023-03-05:
Brilliant! Thank you. Unfortunately, I do have one more problem as shown in the attached image. When I go back into PINN to replace project spaces with those OSes, in its main menu, only the installed OSes are shown, with no tabs. It's definitely got an internet connection because the built in Arora browser can see the internet, but the Reload Repos button doesn't help.

Am I doing something wrong, or is there a known cause of this? I've not got anything important set up yet, so it wouldn't be the end of the world if it's a mystery and I have to start again.

[image]

procount, 2023-03-05:
By using Matt's webpage to determine the partition sizes, you have restricted PINN's view of available OSes to only those that you selected on the webpage. So you need to make some changes to recovery.cmdline.

The easiest thing to do is to restore recovery,cmdline by replacing recovery.cmdline with recovery.cmdline.new.

ajl199, 2023-03-06:
Ah! Now the penny drops about how Matt's page works. Switched recovery.cmdline back and all is well.

Thank you again for doing this.