procount/pinn #711: alt_image_source is only http, redirects to https

macmoritz, 2023-03-15:
Custom alt_image_source URLs are only used if starts with `http://`. 
Found in [file main.cpp line 314](https://github.com/procount/pinn/blob/24ba917afe0281a734ff504effa5c677e664527d/recovery/main.cpp#L308-L317):
```
        else if (strcmp(argv[i], "-alt_image_source") == 0)
        {
             //This could append multiple URLs now
             if (argc > i+1)
             {
                 QString url(argv[i+1]);
                 if (url.startsWith("http://"))
                    downloadRepoUrls << url;
             }
        }
```

makes no sense because the default link `http://pinn.mjh.nz/os_list.json` (http 307, temporary redirect) redirects to https (btw should be http 301 I guess).
This shows that the environment on the device supports https.

Would be great if you can allow `http://` and `https://` for the alt_image_source.
Thanks for your work!

procount, 2023-03-15:
This is for historical reasons.
In order to use https, you need to know the current time.
The Rpi has no RTC, so it gets the time from the first server.
If it doesn't know the time it can only use http.

If you want to use https, you should use a redirect from http to https on the file server.

macmoritz, 2023-03-15:
Makes sense. In fact most web servers have a redirect from http to https nowdays.
Please note the usage of http in the docs.

closed 2023-03-18 by macmoritz