procount/pinn #712: Restore Raspberry Pi OS (64 bit) image with Auto Mount issues

Hiforever, 2023-03-17:
Restore Raspberry Pi OS (64 bit) image with Auto Mount issues.  While auto-mounted storages (be it USB drives or SD Card partitions) still appear in the document explorer, there is nothing in \media\pi (or whatever user) and result in an error.

I am using Raspberry pi 4 and Pinn lite version 3.8.4.   The same problem with other versions of Pinn lite.

procount, 2023-03-17:
This seems to be an extension to issue #447 which I thought was fixed a while ago.
A quick workaround is to unmount all the drives, then delete the /pi folder in /media (or whatever your user name is).
I don't think a reboot is necessary, but can't harm.
Then attach a USB device and the /media/pi folder will be created with the correct permissions.
I shall look into a more permanent fix.

Hiforever, 2023-03-19:
This is resolved.  Thank you very much

closed 2023-03-19 by Hiforever

procount, 2023-03-20:
I have released v3.8.5.3 which should restore the ACL permissions of OS backups properly.