procount/pinn #717: Regarding Kali & RPi 4 8gb

jeff-jm, 2023-03-26:
Hello, I'd first like to thank you for your contribution towards this project.

I am brand new to PINN so I apologize if this is a silly question, but I noticed Kali is not appearing in the list. 

Is it normal for the to fluctuate? Should it reappear soon? I was wondering if it was a problem with my 8gb RPi 4 or something unrelated.

Thanks.

procount, 2023-03-26:
A version of Kali from 2020 is available for the Pi4.
You should find it under the General tab.

The list of available OSes is downloaded from the various servers available on the internet.
If you are not getting a list of OSes, please check your internet connection and make sure you have an IP address showing in the title bar. Do you get any error messages when PINN starts indicating it can't access any of the OS lists?

PINN will filter the OS list to only shows OSes that are suitable for your RPi model, but as you have a Pi4 8 GB, this should not be an issue.

jeff-jm, 2023-03-26:
Hey @procount !

I do not receive any errors when loading the list of OSes from Internet.

I checked again an Kali still seems to be missing from the general tab. The available OSes for me under "General" are:
- Raspberry Pi OS (3 Versions)
- TLXOS (1 Version)
- Lineage (7 Versions)
- twister (1 Version)
- ubuntu (3 Versions)
- Ubuntu Mate (3 Versions)
- gentoo64 (1 Version)
- nspawn64 (1 Version)

I tried installing it manually from a USB to try to see if PINN would recognize it locally, but could not get it to work either.

Let me know if there is anything you recommend I try, thanks again!

procount, 2023-03-26:
You didn't mention Manjaro or RIscOS either in your above list.
This is what I can see:

[image]

and there are a few more beneath that if I scroll down. But you can see that 2 versions of Kali are available.

What is your recovery.cmdline?

Can you post the contents of your debug log? Please see https://github.com/procount/pinn/wiki/Troubleshooting for details on how to capture this, particularly steps 5 or 6.

jeff-jm, 2023-03-26:
Hey @procount , 

Thank you for your continued support.

After looking at your screenshot, I saw your destination drive was a SD card, where as I was trying to use an External USB Drive.

I installed PINN on a USB Drive, and was trying to install additional operating systems on the same USB Drive. After trying an SD Card, all of the OSes in your screenshot above appeared, they only disappear when my destination drive is a USB Drive.

Is this to be expected? Should I try to buy a 64 gb SD Card instead of USB Drive? I apologize for the lack of knowledge as I am new to RPi & Linux distros in general. 

I can try to get you the recovery.cmdline and the debug log if you still think it is relevant, let me know!

THanks so much.

procount, 2023-03-26:
No need to send anything more - I think we found the answer. Apologies for not thinking of it earlier.
Not all OSes are capable of being installed on a USB drive due to the way they reference their partitions, (This is true whether you use it with PINN or not).

The version of Kali I use is quite old now, being from 2020 and is marked as not supporting USB.
It could be that more recent versions can be installed to USB drives, but it would take some time to convert and test them.
You could wait until I can find some time to do that, or if you need an immediate solution a larger SD card would be appropriate.

jeff-jm, 2023-03-26:
@procount ,

Thank you so much, your support was prompt and extensive.

I will go ahead and close this issue as you have clarified everything!

Apologies again for my lack of knowledge and research, wishing you the best.

closed 2023-03-26 by jeff-jm

procount, 2023-03-26:
The good news is that Kali 2023.1 can run from USB. but I still need to see if it will convert to PINN format.

jeff-jm, 2023-03-26:
Awesome!

I will be looking forward to upcoming updates!

I looked around but can't seem to find any donation requests or information regarding the project. I have to ask - is this project run by donations? Or are you a hobbyist/enthusiast yourself just doing it for the community? I am super impressed by the functionality and usability of PINN!

Some other questions I had (if you find time, not important):
- I am just getting into Raspberry Pi things, I mainly want to use it as a tool to get familiarized with Linux distros. 
- I initially planned using PINN on a bootable USB Drive, but as I have discovered it seems to be more simple to use a Micro SD card.
- Are there advantages to using a Micro SD card instead of USB 3.0 Drive? Would an external SSD be a better option?
- Should I avoid using a bootable USB 3.0 and buy a bigger Micro SD Card instead? I only have 30gb. 

Thanks so much again!

procount, 2023-03-30:
Kali64 (2023.1) is now available in PINN.

jeff-jm, 2023-04-01:
> Kali64 (2023.1) is now available in PINN.

Awesome! I now see it is available to install via USB!

I can't thank you enough for all the information you provided about the different booting options. I figured out how to do the "Hybrid Booting" , and am now able to install all the OS's on my USB drive!

Thanks again for everything, I seriously respect and appreciate the work you and others have put into PINN!

procount, 2023-04-01:
You're very welcome.