procount/pinn #718: Firmware not found

Yeag4, 2023-04-07:
Whenever i try to boot my rapberry pi with pinn it says firmware not found, even though I've followed the directions exactly.

procount, 2023-04-07:
Is this your first time using Pinn?
Which midel of RPi are you using?
Which version of Pinn have you installed?
How did you install Pinn?
Did you insrall to SD card or a usb drive?
Can you install other OSes like Raspberry PiOS using RPi Imager?

Yeag4, 2023-04-08:
yes, this is my first time using Pinn
raspberry pi 4 model b
The newest one, don't know the exact name
installed it via sourceforge
installed on to sd card
yes, I can install raspberry pi os for the RPI imager.

procount, 2023-04-09:
You're probably doing something wrong as you have to format the SD card particularly before installing from the zip file, especially if it is a large SD card.
I suggest you try to install from RPi Imager as this is a much easier way.
In Rpi Imager, when you select Choose OS, go to the "Misc utility images" and then the PINN section where you will find the latest version of PINN to install to your SD card.

Yeag4, 2023-04-09:
ok, i will see if that works

Yeag4, 2023-04-09:
that appears to have worked

closed 2023-04-09 by Yeag4