procount/pinn #719: Os not in the list

thostie2002, 2023-04-08:
Can i add my onw os'es that i wanna use but that are not in the list?

ritz-ras1245, 2023-04-08:
Yes, refer to this: https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-create-a-custom-os-version, installation part.

procount, 2023-04-08:
(Sorry, this reply has been languishing in my browser all day as I forgot to hit send!)

Yes, of course.
First you must convert your OS to PINN format. 
Please see https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-create-a-custom-os-version
and there is more of a tutorial on my wiki: https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN.

Once you have your OS converted into a folder with all the installation files, you can install it three ways:
1. Put the folder in the /os folder on your PINN SD card (You need to do this on a fresh installation of PINN before it is booted and resizes the first partition to the minimum size)
2. Get a FAT32 formatted USB drive. Create an 'os' folder on the drive. Put your OS folder in there.
3. Post your OS on the internet and add an 'alt_image_source' parameter in recovery.cmdline to point to it. (This uses a slightly differently formatted os_list.v3.json file rather than the os.json file.