procount/pinn #721: Getting an error when building pinn

ToberoCat, 2023-04-15:
Hi. I've tried to build pinn on my own, so I can add the tsc-2007 touchscreen driver. 
I setup a docker container with the build instructions in the readme, but when I run the build.sh, I get following error:
```
tput: No value for $TERM and no -T specified
tput: No value for $TERM and no -T specified
You must install 'python' on your build machine
make: *** [support/dependencies/dependencies.mk:28: core-dependencies] Error 1
```

I already tried running `sudo apt-get install python3 python2` and even confirmed the python installation with `python --version`. I still get this error and I can't do anything about it.

So my question is, if there's someone who can help me with it. If someone has a build for the tsc-2007 touchscreen, I'd take that too.

procount, 2023-04-15:
Yeah, you have to make sure you have all the right packages and there are several patches for the buildroot kernel/packages to get it to build correctly. The pre-requisites assume you are building on Ubuntu 20.04, so it may not work if you build on any other distro without a bit of tweaking.

I use the free VMWare Player (version 15) to install a virtual machine of Ubuntu 20.04.1 (64-bit) as my development environment. You'll need to allocate it about 40GB of HDD space to compile PINN.

ToberoCat, 2023-04-15:
Okay. Thanks. I'll try to use ubuntu 20.04 instead of 22

procount, 2023-04-16:
[Downloads.zip](https://github.com/procount/pinn/files/11242673/Downloads.zip)
You could try these.

ToberoCat, 2023-04-16:
I replaced the files on the sd card with the ones you sent in the zip, but the touchscreen still doesn't work.

I also tried running the docker container on ubuntu20.04, but this didn't fix the issue.

Here are the outputs:
```
tput: No value for $TERM and no -T specified
tput: No value for $TERM and no -T specified
You must install 'cpio' on your build machine
You must install 'python' on your build machine
make: *** [support/dependencies/dependencies.mk:28: core-dependencies] Error 1
```

Running apt install cpio resolved the issue with cpio, but python still persists, with python and python2 installed. Even using an alias to map python3 to python didn't fix it.

Docker file: https://pastebin.com/pG0Y7ytk

procount, 2023-04-16:
You probably have to add an appropriate dtoverlay in config.txt to get the touchscreen drivers to load, but I'm not familiar with that touchscreen. 

It looks like the version of Ubuntu 20.04 you download for Docker is a very minimal version with a different set of installed packages to the full version. I don't run docker so I can't really help there. When RPF updated their Ubuntu development image for NOOBS from 14.04 to 20.04, I had to follow suit for PINN to get it to build at all. I'm not able to support other development environments, other than running Ubuntu 20.04 on a PC either directly or under Vmware player.

I have python 2.7.18 and 3.8.10 installed, if that helps. Python is linked to python2.

procount, 2023-04-16:
https://forums.raspberrypi.com/viewtopic.php?t=224018

ToberoCat, 2023-04-20:
Thanks. I've finally fixed the issue with docker file needing python. I'll be making a pull request that adds the necessary files so people can build PINN using docker too.

Now I'll be figuring out how to include the drivers for the display, as the overlay didn't seem to work

ToberoCat, 2023-04-23:
For some weird reason I can't build pinn anymore after using `make clean`. The error points to me that it can't reach github and the sources.buildroot.net returns 404. 
I already checked my wifi, but I'm able to contact github and sources.buildroot.net.

Do you maybe know what's causing the issue?

```
>>> linux b9c9f8fc69aaba236169798c18dc1590fccf7acd Downloading
Doing shallow clone
Cloning into bare repository 'linux-b9c9f8fc69aaba236169798c18dc1590fccf7acd'...
Looking up github.com ... done.
Connecting to github.com (port 9418) ... fatal: unable to connect to github.com:
github.com[0: 140.82.121.4]: errno=Connection refused
github.com[1: ::ffff:140.82.121.4]: errno=Connection refused
Shallow clone failed, falling back to doing a full clone
Doing full clone
Cloning into bare repository 'linux-b9c9f8fc69aaba236169798c18dc1590fccf7acd'...
Looking up github.com ... done.
Connecting to github.com (port 9418) ... fatal: unable to connect to github.com:
github.com[0: 140.82.121.4]: errno=Connection refused
github.com[1: ::ffff:140.82.121.4]: errno=Connection refused
--2023-04-23 11:46:51--  http://sources.buildroot.net/linux-b9c9f8fc69aaba236169798c18dc1590fccf7acd.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 172.67.72.56, 104.26.1.37, 104.26.0.37, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|172.67.72.56|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-04-23 11:46:52 ERROR 404: Not Found.
make: *** [package/pkg-generic.mk:75: /home/dev/pinn/buildroot/output/build/linux-b9c9f8fc69aaba236169798c18dc1590fccf7acd/.stamp_downloaded] Error 1
```

procount, 2023-04-23:
I sometimes have problems with buildroot downloading the kernel files.
I  end up downloading the linux-....tar.gz file directly from the rp github and putting it in buildroot/dl.
Maybe a good idea to persist that folder in a docker volume to avoid buildroot downloading it each time.

ToberoCat, 2023-04-25:
Where can I find the files? When I searched for the one that was failing, I haven't found a result

procount, 2023-04-25:
FIrstly, what version of PINN are you trying to build?
You seem to be trying to use linux commit id b9c9f8fc69aaba236169798c18dc1590fccf7acd which is not in the latest version of PINN.
PINN uses commit id dc6771425e9604650d1d57f7c69948be405f59a5 from the raspberry pi github, which is probably not available from sources.buildroot.net.
If you are trying to use the latest version of PINN, please check your .config file is correct and set to the correct linux commit.

I download it from: https://github.com/raspberrypi/linux/archive/dc6771425e9604650d1d57f7c69948be405f59a5.zip, or alternatively there is a .tar.gz version. Just put it in buildroot/dl before building.

By the way, PINN builds using ccache, so after the first build, subsequent builds should be faster, but if you do not persist the ~/.buildroot-ccache folder and buildroot/dl folder (maybe also buildroot/output) between running instances of docker, you may find it takes a long time to build each time.

ToberoCat, 2023-04-26:
Yeah, I tried to build the latest. I'll try that when I'm home.
Thank you

procount, 2023-04-28:
I've not seen that one before. 
I'm guessing that dl-wrapper is some buildroot tool to do the download and it is missing a backend package that's needed to download binutils. 
This is the file that mine downloads: http://ftp.gnu.org/pub/gnu/binutils/binutils-2.24.tar.bz2
So maybe it's an FTP issue or bz2 issue?
Try downloading that file manually as before.

ToberoCat, 2023-05-06:
Hi. It's me again.
I've finally managed to get buildroot running. I've ran it for half a day compiling and linking linux.
Now, it looks like there is an issue with creating the image.
```>>> linux dc6771425e9604650d1d57f7c69948be405f59a5 Installing to images directory
cp /home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5/arch/arm/boot/zImage /home/dev/pinn/buildroot/output/images
# dtbs moved from arch/<ARCH>/boot to arch/<ARCH>/boot/dts since 3.8-rc1
cp /home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5/arch/arm/boot/dts/bcm27*.dtb /home/dev/pinn/buildroot/output/images/
mkdir -p /home/dev/pinn/buildroot/output/images/overlays/
cp /home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5/arch/arm/boot/dts/overlays/*.dtbo /home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5/arch/arm/boot/dts/overlays/README /home/dev/pinn/buildroot/output/images/overlays/
tput: No value for $TERM and no -T specified
tput: No value for $TERM and no -T specified
>>> ncurses 5.9 Installing to target
mkdir -p /home/dev/pinn/buildroot/output/target/usr/lib
for lib in libncurses; do cp -dpf /home/dev/pinn/buildroot/output/build/ncurses-5.9/lib/${lib}.so* /home/dev/pinn/buildroot/output/target/usr/lib/; done
ln -snf /usr/share/terminfo /home/dev/pinn/buildroot/output/target/usr/lib/terminfo
mkdir -p /home/dev/pinn/buildroot/output/target/usr/share/terminfo/x
cp -dpf /home/dev/pinn/buildroot/output/host/usr/arm-buildroot-linux-uclibcgnueabi/sysroot/usr/share/terminfo/x/xterm /home/dev/pinn/buildroot/output/target/usr/share/terminfo/x
cp: cannot stat '/home/dev/pinn/buildroot/output/host/usr/arm-buildroot-linux-uclibcgnueabi/sysroot/usr/share/terminfo/x/xterm': No such file or directory
make: *** [package/pkg-generic.mk:236: /home/dev/pinn/buildroot/output/build/ncurses-5.9/.stamp_target_installed] Error 1
rm -f /home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5/.stamp_built
rm -f /home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5/.stamp_staging_installed
rm -f /home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5/.stamp_target_installed
rm -f /home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5/.stamp_images_installed
rm -f /home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5/.stamp_host_installed
rm -f /home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5/.stamp_configured
/usr/bin/make -j1  HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" silentoldconfig
make[1]: Entering directory '/home/dev/pinn/buildroot'
BR2_DEFCONFIG='' KCONFIG_AUTOCONFIG=/home/dev/pinn/buildroot/output/build/buildroot-config/auto.conf KCONFIG_AUTOHEADER=/home/dev/pinn/buildroot/output/build/buildroot-config/autoconf.h KCONFIG_TRISTATE=/home/dev/pinn/buildroot/output/build/buildroot-config/tristate.config BR2_CONFIG=/home/dev/pinn/buildroot/.config BR2_EXTERNAL=support/dummy-external /home/dev/pinn/buildroot/output/build/buildroot-config/conf --silentoldconfig Config.in
make[1]: Leaving directory '/home/dev/pinn/buildroot'
>>> linux dc6771425e9604650d1d57f7c69948be405f59a5 Configuring
/usr/bin/install -m 0644 "kernelconfig-recovery.armv7l" /home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5/arch/arm/configs/buildroot_defconfig
PATH="/home/dev/pinn/buildroot/output/host/bin:/home/dev/pinn/buildroot/output/host/sbin:/home/dev/pinn/buildroot/output/host/usr/bin:/home/dev/pinn/buildroot/output/host/usr/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" /usr/bin/make -j1 HOSTCC="/home/dev/pinn/buildroot/output/host/usr/bin/ccache /usr/bin/gcc -O2 -I/home/dev/pinn/buildroot/output/host/usr/include -L/home/dev/pinn/buildroot/output/host/lib -L/home/dev/pinn/buildroot/output/host/usr/lib -Wl,-rpath,/home/dev/pinn/buildroot/output/host/usr/lib" ARCH=arm INSTALL_MOD_PATH=/home/dev/pinn/buildroot/output/target CROSS_COMPILE="/home/dev/pinn/buildroot/output/host/usr/bin/ccache /home/dev/pinn/buildroot/output/host/usr/bin/arm-buildroot-linux-uclibcgnueabi-" DEPMOD=/home/dev/pinn/buildroot/output/host/sbin/depmod -C /home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5 buildroot_defconfig
make[1]: Entering directory '/home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5'
#
# configuration written to .config
#
make[1]: Leaving directory '/home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5'
rm /home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5/arch/arm/configs/buildroot_defconfig
/usr/bin/sed -i -e "/\\<CONFIG_AEABI\\>/d" /home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5/.config
echo 'CONFIG_AEABI=y' >> /home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5/.config
# As the kernel gets compiled before root filesystems are
# built, we create a fake cpio file. It'll be
# replaced later by the real cpio archive, and the kernel will be
# rebuilt using the linux-rebuild-with-initramfs target.
/usr/bin/sed -i -e "/\\<CONFIG_DEVTMPFS\\>/d" /home/dev/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5/.config
[... cut ...]
fs/btrfs/tree-checker.c:1078:9: warning: missing braces around initializer [-Wmissing-braces]
  struct btrfs_root_item ri = { 0 };
         ^
fs/btrfs/tree-checker.c:1078:9: warning: (near initialization for 'ri.inode') [-Wmissing-braces]
cp: cannot stat 'output/images/rootfs.squashfs': No such file or directory
```

So essentially, this error says that it searches the rootfs.squashfs file. I wanted to ask if you encountered it. If yes, how did you solve it?
My guess is that I have to download it, like I did with the other packages, but I'm not able to find such a file on the web

procount, 2023-05-06:
Rootfs.squashfs is the root filesystem that it is trying to build. It seems to be failing on building the fake cpio file. Is cpio installed?

ToberoCat, 2023-05-06:
Yep, cpio got installed using apt.

procount, 2023-05-06:
I've not encountered this problem before. Are there any prior errors in the logs?

ToberoCat, 2023-05-07:
I've now searched through the build log file and only thing that was mentioning a error was this section.

I'm not quite sure why tput is complaining, as I've set the TERM env variable.
I've also checked the directory `/home/dev/pinn/buildroot/output/host/usr/arm-buildroot-linux-uclibcgnueabi/sysroot/usr/share/terminfo/`, but it doesn't seem like folder `x` has been created.
```
tput: No value for $TERM and no -T specified
tput: No value for $TERM and no -T specified
>>> ncurses 5.9 Installing to target
mkdir -p /home/dev/pinn/buildroot/output/target/usr/lib
for lib in libncurses; do cp -dpf /home/dev/pinn/buildroot/output/build/ncurses-5.9/lib/${lib}.so* /home/dev/pinn/buildroot/output/target/usr/lib/; done
ln -snf /usr/share/terminfo /home/dev/pinn/buildroot/output/target/usr/lib/terminfo
mkdir -p /home/dev/pinn/buildroot/output/target/usr/share/terminfo/x
cp -dpf /home/dev/pinn/buildroot/output/host/usr/arm-buildroot-linux-uclibcgnueabi/sysroot/usr/share/terminfo/x/xterm /home/dev/pinn/buildroot/output/target/usr/share/terminfo/x
cp: cannot stat '/home/dev/pinn/buildroot/output/host/usr/arm-buildroot-linux-uclibcgnueabi/sysroot/usr/share/terminfo/x/xterm': No such file or directory
make: *** [package/pkg-generic.mk:236: /home/dev/pinn/buildroot/output/build/ncurses-5.9/.stamp_target_installed] Error 1
```

procount, 2023-05-07:
$ echo $TERM
xterm-256color