procount/pinn #723: Can someone wiser tell me why my os won't install

alreadyconfused, 2023-04-27:
I can tell that there's a lot going wrong here but don't really know where to start. 
Here's the log file output that i got from savelog :

`DEBUG
=====
Keyboard found: "/dev/input/event0" 
3.8.6 
Board revision is  10494163 
Found recovery.rfs at "mmcblk0p1" 
PINN drive: "/dev/mmcblk0" 
mount: /dev/mmcblk0 is already mounted or /mnt busy
Simulate OK 
CecListener thread started 
Looking for  "/mnt/cec_keys.json" 
CecListener done initializing 
Loading Mappings mappings from  ":/cec_keys.json" 
New =  mainwindow ,  Main Menu 
setWindow  "mainwindow" 
setMenu  "Main Menu" 
Starting dbus 
Starting dhcpcd 
Default language is  "en" 
Default keyboard layout is  "gb" 
Default display mode is  "0" 
Error opening file: "/settings/installed_os.json" 
Number of bootables =  0 
New drive selected: "mmcblk0" 
cp: can't stat '/mnt/os/RaspiOS_Full_armhf/terms': No such file or directory
rm: can't remove '/tmp/progress': No such file or directory
partition map: QMap((1, PartitionInfo(0x20df678) ) ( 2 ,  PartitionInfo(0x207df60) ) ( 5 ,  PartitionInfo(0x2119cc0) ) ( 6 ,  PartitionInfo(0x203d198) ) ( 7 ,  PartitionInfo(0x214f5e0) ) )  
New partition table: 
"2048,1927640,0E
1929688,29186600,E
0,0
0,0
1933312,65534,L
1998848,524286,0c
2523136,28593152,83
" 
sfdisk done, output: "/sbin/sfdisk: Checking that no-one is using this disk right now ...
/sbin/sfdisk: OK
/sbin/sfdisk: Warning: extended partition does not start at a cylinder boundary.
DOS and Linux will interpret the contents differently.
/sbin/sfdisk: Warning: partition 1 does not end at a cylinder boundary
/sbin/sfdisk: Warning: partition 2 does not start at a cylinder boundary
/sbin/sfdisk: Warning: partition 5 does not end at a cylinder boundary
/sbin/sfdisk: Warning: partition [6] does not start at a cylinder boundary
/sbin/sfdisk: Warning: partition 6 does not end at a cylinder boundary
/sbin/sfdisk: Warning: partition [7] does not start at a cylinder boundary
/sbin/sfdisk: Warning: no primary partition is marked bootable (active)
This does not matter for LILO, but the DOS MBR will not boot this disk.

Disk /dev/mmcblk0: 486192 cylinders, 4 heads, 16 sectors/track
Old situation:
Units: sectors of 512 bytes, counting from 0

   Device Boot    Start       End   #sectors  Id  System
/dev/mmcblk0p1          2048   1929687    1927640   e  W95 FAT16 (LBA)
/dev/mmcblk0p2       1929688  31116287   29186600   5  Extended
/dev/mmcblk0p3             0         -          0   0  Empty
/dev/mmcblk0p4             0         -          0   0  Empty
/dev/mmcblk0p5       1933312   1998845      65534  83  Linux
/dev/mmcblk0p6       1998848   2523133     524286   c  W95 FAT32 (LBA)
/dev/mmcblk0p7       2523136  31116287   28593152  83  Linux
New situation:
Units: sectors of 512 bytes, counting from 0

   Device Boot    Start       End   #sectors  Id  System
/dev/mmcblk0p1          2048   1929687    1927640   e  W95 FAT16 (LBA)
/dev/mmcblk0p2       1929688  31116287   29186600   5  Extended
[... 175 lines cut ...]
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/ram15: 32 MiB, 33554432 bytes, 65536 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disk /dev/mmcblk0: 14.9 GiB, 15931539456 bytes, 31116288 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x67ed0df0

Device         Boot   Start      End  Sectors   Size Id Type
/dev/mmcblk0p1         2048  1929687  1927640 941.2M  e W95 FAT16 (LBA)
/dev/mmcblk0p2      1929688 31116287 29186600  13.9G  5 Extended
/dev/mmcblk0p5      1933312  1998845    65534    32M 83 Linux
/dev/mmcblk0p6      1998848  2523133   524286   256M  c W95 FAT32 (LBA)
/dev/mmcblk0p7      2523136 31116287 28593152  13.6G 83 Linux

MEDIA
=====

`

alreadyconfused, 2023-04-27:
Just to clarify: i copied the RaspiOS_Full_armhf folder from NOOBS and then replaced the boot and root files with my custom configured legacy rpi lite os files. Probably not the most intuitive thing to do

alreadyconfused, 2023-04-27:
Also when looking at the original tar.xz files from NOOBS' RaspiOS_Full_armhf folder, all the directories and files inside the .xz files are at the root. When i imaged the my os' sd card, then used the tar command in ubuntu to make the tar.xz files, it showed that they also were in the root of the file, but when opening the same files in windows with winrar, the uncompressed tar file is in the root of the tar.xz file. So does all the necessary files and directories need to be in the root of the tar.xz file, and if so how do i exactly do that?

procount, 2023-04-28:
If you have a customised RaspioOS **LITE** OS, would it make more sense to start off with a NOOBS RaspiOS LITE installation?

You might want to consider renaming it's folder from "Raspios Full...." to "VideoLooper" to distinguish it and to be consistent with the rest of the naming.

> Executing: "sh -o pipefail -c "cat '/mnt/os/RaspiOS_Full_armhf/boot.tar.xz' | tee /tmp/fifo | xz -dcT 0 | bsdtar -xf - -C /mnt2  --no-same-owner  | sha512sum /tmp/fifo > /tmp/sha1.out.txt"" 
"xz: (stdin): File format not recognized
" 

There seems to be some issues with your tar.xz files. have you followed the instructions/tutorials I provided in the README_PINN.md and on the Wiki? You probably need to install the prescribed bsdtar version

lurch, 2023-05-03:
> when opening the same files in windows with winrar, the uncompressed tar file was inside the root of the tar.xz file

I think that's just a quirk of some Windows compression software - they're not used to the "double wrapper" commonly used by Linux archives, so they show a `.tar` file "inside" the `.xz` file, when that's not _really_ the case (depending on how you look at things!).

closed 2024-08-13 by alreadyconfused