procount/pinn #727: DSI touchscreen quirk to get detected as TAP

fthiebolt, 2023-05-03:
Hello proCount,

I detected that, when having a DSI display connected to my Raspberry Pi, when rebooting the system, it behaves like if someone touched the screen (i.e see 'TAP detected' below). 
Is there a way to 'flush' the touchscreen input driver to allow automatic reboot on the installed system ?

root@recovery:~# cat /var/log/debug 
3.8.6 
Board revision is  10494163 
Could not read calibration: "/etc/pointercal" 
Found recovery.rfs at "mmcblk0p1" 
PINN drive: "/dev/mmcblk0" 
mount: /dev/mmcblk0 is already mounted or /mnt busy
Simulate OK 
CecListener thread started 
Looking for  "/mnt/cec_keys.json" 
Not in HDMI mode 
Loading Mappings mappings from  ":/cec_keys.json" 
Tap detected 
New =  mainwindow ,  Main Menu 
setWindow  "mainwindow" 
setMenu  "Main Menu" 
Processing Selection:  "Raspbian,waitnetwork" 
Starting dbus 
Starting dhcpcd 
Default language is  "en" 
Default keyboard layout is  "fr" 
Default display mode is  "0" 
Number of bootables =  1 
Network up in 11.855 seconds 
current date is  QDate("Thu Jan 1 1970") 
Requesting current time 
Skipping self update check 
downloadRepoList:  "" 
Waiting for OSes... 4  Got  1 
New drive selected: "mmcblk0" 
Time set to:  QDateTime("Wed May 3 08:52:13 2023") 
Number of bootables =  1 
Waiting for OSes... 4  Got  5 
Selecting OSes... 
   "Raspbian" 
Selecting installed OSes... 
"Checking PINN vs Raspbian" 
"Checking Raspbian vs Raspbian" 
found 

Thanks for the excellent work :)

procount, 2023-05-04:
I guess this is not the official Raspberry PI 7" DSI touchscreen because mine behaves ok, and I have taken steps in PINN to avoid such ghost taps. What display is it, and more importantly, which touchscreen driver does it use?

fthiebolt, 2023-05-04:
Hello and thanks for your quick answer.
At least, my university bought them as official 7 inch DSI displays ... and i already tested on 3 others (identicals) with the same behaviour.

root@rpi3-dsi-install[~] DISPLAY=:0 xinput 
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ generic ft5x06 (79)                       id=6    [slave  pointer  (2)]
⎜   ↳ vc4-hdmi                                  id=7    [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ vc4-hdmi                                  id=8    [slave  keyboard (3)]

For completness:
root@rpi3-dsi-install[~] lsmod
Module                  Size  Used by
xt_conntrack           16384  1
nft_chain_nat          16384  3
xt_MASQUERADE          16384  1
nf_nat                 49152  2 xt_MASQUERADE,nft_chain_nat
nf_conntrack_netlink    45056  0
nf_conntrack          135168  4 xt_MASQUERADE,nf_conntrack_netlink,xt_conntrack,nf_nat
nf_defrag_ipv6         20480  1 nf_conntrack
nf_defrag_ipv4         16384  1 nf_conntrack
rfcomm                 49152  4
xt_addrtype            16384  2
nft_compat             20480  4
nf_tables             225280  57 nft_compat,nft_chain_nat
nfnetlink              20480  4 nft_compat,nf_conntrack_netlink,nf_tables
br_netfilter           32768  0
bridge                184320  1 br_netfilter
stp                    16384  1 bridge
llc                    16384  2 bridge,stp
cmac                   16384  3
algif_hash             16384  1
aes_arm_bs             24576  2
crypto_simd            16384  1 aes_arm_bs
cryptd                 24576  2 crypto_simd
algif_skcipher         16384  1
af_alg                 28672  6 algif_hash,algif_skcipher
bnep                   20480  2
hci_uart               40960  1
btbcm                  20480  1 hci_uart
bluetooth             503808  31 hci_uart,bnep,btbcm,rfcomm
ecdh_generic           16384  2 bluetooth
ecc                    40960  1 ecdh_generic
overlay               118784  0
joydev                 20480  0
tc358762               16384  0
rpi_panel_attiny_regulator    16384  5
spidev                 20480  0
regmap_i2c             16384  1 rpi_panel_attiny_regulator
edt_ft5x06             20480  0
vc4                   315392  6
brcmfmac              335872  0
snd_soc_hdmi_codec     16384  1
drm_display_helper     16384  1 vc4
brcmutil               20480  1 brcmfmac
cec                    49152  1 vc4
drm_dma_helper         20480  1 vc4
drm_kms_helper        188416  5 drm_dma_helper,vc4,tc358762
sha256_generic         16384  0
snd_soc_core          253952  2 vc4,snd_soc_hdmi_codec
cfg80211              811008  1 brcmfmac
snd_compress           20480  1 snd_soc_core
snd_pcm_dmaengine      20480  1 snd_soc_core
bcm2835_codec          45056  0
syscopyarea            16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
i2c_mux_pinctrl        16384  0
rfkill                 32768  6 bluetooth,cfg80211
i2c_mux                16384  1 i2c_mux_pinctrl
raspberrypi_hwmon      16384  0
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
bcm2835_v4l2           45056  0
v4l2_mem2mem           40960  1 bcm2835_codec
bcm2835_isp            32768  0
bcm2835_mmal_vchiq     36864  3 bcm2835_isp,bcm2835_codec,bcm2835_v4l2
videobuf2_vmalloc      16384  1 bcm2835_v4l2
videobuf2_dma_contig    20480  2 bcm2835_isp,bcm2835_codec
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_vmalloc
videobuf2_v4l2         32768  4 bcm2835_isp,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
videobuf2_common       65536  8 bcm2835_isp,bcm2835_codec,videobuf2_dma_contig,videobuf2_vmalloc,videobuf2_memops,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
videodev              266240  6 bcm2835_isp,bcm2835_codec,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
snd_bcm2835            24576  1
spi_bcm2835            20480  0
snd_pcm               118784  5 snd_compress,snd_pcm_dmaengine,snd_soc_hdmi_codec,snd_bcm2835,snd_soc_core
snd_timer              36864  1 snd_pcm
i2c_bcm2835            16384  0
snd                    94208  8 snd_compress,snd_soc_hdmi_codec,snd_timer,snd_bcm2835,snd_soc_core,snd_pcm
vc_sm_cma              32768  2 bcm2835_isp,bcm2835_mmal_vchiq
mc                     53248  6 bcm2835_isp,bcm2835_codec,videobuf2_common,videodev,v4l2_mem2mem,videobuf2_v4l2
panel_simple           81920  0
uio_pdrv_genirq        16384  0
uio                    24576  1 uio_pdrv_genirq
fixed                  16384  1
drm                   544768  11 drm_dma_helper,vc4,panel_simple,tc358762,drm_display_helper,drm_kms_helper
i2c_dev                16384  0
fuse                  131072  3
drm_panel_orientation_quirks    16384  1 drm
backlight              20480  2 rpi_panel_attiny_regulator,drm
ip_tables              28672  0
x_tables               36864  5 ip_tables,nft_compat,xt_MASQUERADE,xt_addrtype,xt_conntrack
ipv6                  520192  44 br_netfilter,bridge

procount, 2023-05-04:
Oh, that is surprising as it does seem to be the same.
Hmm. If you're agreeable, I think I'll have to create a debug version for you to test in order to see what is going on.

fthiebolt, 2023-05-04:
of course ;)

procount, 2023-05-04:
OK. Bear with me, bit busy right now, but will prepare one for you.