procount/pinn #728: Switch to a supported browser

mattruzzi, 2023-05-08:
Can you switch from aurora (which is discontinued and doesn't properly support many modern sites) to a newer browser that is still supported?

lurch, 2023-05-08:
It'd need to be a browser that supports QT Embedded (IIRC PINN still uses an older version of QT Embedded?), and is able to run in the very cut-down (BuildRoot) Linux environment used by PINN. Do you have any suggestions?

mattruzzi, 2023-05-08:
The Endorphin web browser is a fork of aurora which is still in development. The [Falkon](https://www.falkon.org/) browser is another browser that is based on qt. I don't really know how qt embedded works so I don't know if either of them would actually work.

lurch, 2023-05-08:
> The Endorphin web browser is a fork of aurora which is still in development.

Haha, one of the names at the bottom of https://endorphinbrowser.github.io/ looks familiar :joy:  I've not contributed directly to Endorphin, but I did contribute to Arora a long time ago.

mattruzzi, 2023-05-09:
Would it be possible to use Falkon browser? It looks like it is a lot more up to date.

lurch, 2023-05-09:
Just out of curiosity, why are you so reliant on the web-browser built into PINN? Perhaps one of the "kiosk mode" OSes for Raspberry Pi would suit your needs better?

TheUnkn0wnHack3r, 2023-09-23:
> Just out of curiosity, why are you so reliant on the web-browser built into PINN? Perhaps one of the "kiosk mode" OSes for Raspberry Pi would suit your needs better?

My best guess is that the op needs to upload the logs from PINN (which are stored in /tmp/debug) to a external website. However, the default browser (as of right now) doesn't load websites properly (just plain html, no css and probably no javascript either)