procount/pinn #732: Clone Feature Buggy

casedup, 2023-06-21:
I seemed to accidently got the clone feature to work once and since then it's been throwing me a cannot mount destination /tmp/dst. No idea why it is throwing the error and I don't see much in the documentation regarding the way it works.

procount, 2023-06-21:
What OSes are on the card you are trying to clone?
What does the log say?

casedup, 2023-06-21:
Trying to clone LibreElec, Raspian, Recalbox to a new SD Card. Tried the clone tool within raspian with both cards none live in USB SD Card Reader. All throw a mount error. How exactly do you view the log? I found a clone option in my EASE software attempting that now.

procount, 2023-06-21:
See https://github.com/procount/pinn/wiki/Troubleshooting to view the log.

The clone function works exactly the same as the SD card copier in Raspbian. As you have that installed, maybe you could try that instead if you have the desktop version?

Since a straight 'dd' copy can have issues with cards of different sizes, the clone feature recreates the partition layout on the target device and then tries to copy (rsync) the files from each source partition to their respective target partition. It can fail if the OSes use unsupported filesystems. 

I don't use this feature and haven't tested it in years. So I just tried it and it failed after a while, so I will have to investigate further.

procount, 2023-06-21:
Whilst I'm not getting the same mount error, the clone is failing even though it claims to be successful. The difference may be just due to timing .
Fortunately, I think I've found the issue which is due to interaction with another feature which is making the USB drive read-only. I will look at fixing this. Thanks for reporting it.

casedup, 2023-06-21:
i was thinking the same... some kind of permission error which makes no sense when the drive was properly formatted. i even used the recommended sd card formatter and added a label, still nothing. fyi the EASE software seemed to finish without throwing a error. on another side note i accidently got it to work once, so i know clone multiple partitions is possible. also tried cloning raspian running live to sd card and wrote perfectly so im guessing the bug is in the way the partitions are handled.

casedup, 2023-06-21:
yeah ive tried using raspian desktop to initiate the clone, both with the SD with the multiple partitions running as the OS and also tried with a SD running a standalone raspian desktop with both source and target SD cards inside the reader. all throw a mounting error.

procount, 2023-06-21:
No need to reformat the target before cloning, because it will be repartitioned and reformatted anyway, destroying what you have already done.
I am surprised the Raspbian SD Card Copied also throws a mount error.
I thought I understood why the target was going read-only, but it needs some more investigation - may take a while.

casedup, 2023-06-21:
the raspian sd card software works perfectly when dealing with a single partition, only the multiple partitions i'm getting the mount error on.

procount, 2023-06-21:
Ah, I don't think it is set up for that.