procount/pinn #736: How to download custom OS

Sweeistaken, 2023-07-11:
Hi.
I got PINN for multibooting a few operating systems, but I specifically wanted to have Debian ARM port on the multiboot.
well the download button is greyed out and there is no Debian maintained by this community. is there a way I can download a custom OS from the internet using PINN?

procount, 2023-07-11:
Hi, There was Debian version some time ago, but I haven't updated it for a while. I can do that, though. Which model of Pi?
Could you not use RaspberryPi OS? It is basically Debian with a custom desktop.
Please read https://github.com/procount/pinn/blob/master/README_PINN.md#download as there are some conditions to be met before download is available (e.g. have  USB stick with an 'os' folder in it).
In the above link (and on my wiki) there are instructions on how to convert an OS to PINN format yourself if you have a custom one you need converting.

Sweeistaken, 2023-07-11:
> 

I am using the Raspberry pi 2B. and no, because Raspbian has gave me some problems and limitations with it's repositories and other stuff and I am looking for an alternative

closed 2023-07-15 by Sweeistaken