procount/pinn #738: Debian Bookworm for Pi2

Sweeistaken, 2023-07-15:


procount, 2023-07-17:
I'm struggling a bit with this one. Do you need a desktop version or would cmdline suffice? You can always install a desktop afterwards using tasksel.

Sweeistaken, 2023-07-17:
@procount the cmdline version is Okay, I also have problems finding a desktop version of it.

procount, 2023-07-17:
I have a cmdline version installing, but for some reason adding a desktop doesn't allow it to boot to the desktop after re-installing with PINN. I just need to add the proper logo and slideshow etc.
Once the cmdline version is installed, do an `apt update` and then use `tasksel` to install a desktop of your choice.

Sweeistaken, 2023-07-17:
Ok. Thanks for the tips because I completely forgot about tasksel, so I tried using debconf instead 😂.

Sweeistaken, 2023-07-19:
@procount I installed the MATE desktop on debian, but when LightDM starts, it just shows the underline cursor at the top left of the screen and after 1 second it is just a black screen.

procount, 2023-07-19:
I get that too with the few I tried. :(

Sweeistaken, 2023-07-20:
Yeah, I tried using the config.txt from my Ubuntu mate edit, but still no luck

closed 2023-07-29 by Sweeistaken