procount/pinn #740: Very slow download/install of images lately

89carrot, 2023-07-20:
Good Day,

I've been trying to install directly or download for offline both LibreElec and Batocera for a dual boot setup.

LibreElec in particular seems extremely slow where the download just stops.

Would there be an alternate source for these files or is there an issue with the repository?

Thank you

grozzle, 2023-08-07:
Same. I'm getting an estimate of about two days to download.

procount, 2023-08-07:
Both are hosted by their respective maintainers so I have no control over that.

procount, 2023-09-19:
I am not experiencing these issues.
Most of the OSes are stored on Sourceforge. It could be that it is choosing the wrong (i.e. slow) mirrors for your location. Which country are you in? I am thinking about adding a feature to allow the selection of mirrors, if that is possible, but that may be a way off.
PINN can download the "images" to a USB for local installation (see README_PINN.md), but it will use the same mirrors so it probably won't be any quicker. 
Alternatively you can download the OSes from Sourceforge directly (e.g. https://sourceforge.net/projects/pinn/files/os/) to a suitably formatted USB drive using your browser (where you can select an appropriate mirror) , and then install them locally from there.

mattruzzi, 2023-09-20:
Could you also host the OS downloads on GitHub releases as an alternate download option?

lucamoura, 2024-02-11:
I'm having a similar issue. At the moment 'Wrintten' appears initially, the speed starts around 9.5MB/sec but gradually decreases to 0.9 and keeps decreasing until it stops. I have PINN OS installed on an SSD via USB 3.0 on a Raspberry Pi 4 2GB. Simply, from 40%, the remaining time just keeps increasing.
[image]

lucamoura, 2024-02-11:
[image]

darksharkx, 2024-12-29:
I’m getting 1.5 MB/sec on a 500Mbps connection.
Adding mirrors would be greatly appreciated