procount/pinn #741: Backup button won't enable

ShadowEO, 2023-07-21:
I am assuming that I'm doing something extremely wrong, but I'm trying to back up a copy of the "raspios_arm64_lite" OS to a USB flash drive. The button just stays disabled, but I see in the JSON files that it does indeed support backup. I've tried it both online and offline to no avail. Is it because I'm using a FAT32 formatted flash drive, as opposed to an external HDD?

procount, 2023-07-21:
Please see: https://github.com/procount/pinn/blob/master/README_PINN.md#backup
Did you create an `os` folder on your usb drive?

ShadowEO, 2023-07-22:
.... omg... please feel free to just delete this ticket.... I feel like a moron, I knew I was missing a step..

closed 2023-07-22 by ShadowEO