procount/pinn #743: Install Multiple OS's to both MicroSD and USB

V01D-ST3R, 2023-07-28:
Hi, is it possible to install Pinn to the MicroSD and install OS's to both the MicroSD and USB or can I only have Pinn installed on the MicroSD and all OS's go on the USB? I have installed Pinn along with 2 OS's to the MicroSD, and after reboot I have plugged in a USB and selected 6 OS's to install to the USB but I get a warning that the currently installed OS's will be wiped(I'm assuming the MicroSD will be wiped or Pinn re-installed). Is there a solution or is it not possible?

procount, 2023-07-28:
Noobs/pinn were originally designed as a one-shot installer so there is no provision to add OSes later or install to different drives because it expands the OSes to fill all the drive.
With PINN you can now replace an OS individually, and install blank ProjectSpaces which can later be replaced with another compatible OS.

You could get another microSD card and install PINN  to it to allow you to install OSes to your USB.
Or you could just install PINN to your USB drive and install/boot from there. Then there is no need for a microsd card.

V01D-ST3R, 2023-07-28:
How would I go about flashing Pinn to a USB? I flashed Pinn to a USB with Raspberry Pi imager but on boot it shows a message again about wiping the previous OS's and when I press cancel it displays the OS's from the MicroSD. I tried changing the "root=/" in recovery.cmdline to "root=/dev/sda2" but that just fails to boot. I have also tried the "Pinn System Resize" webpage which has an option for USB drive and outputs a customized recovery.cmdline but I end up with the first issue of Pinn wanting to wipe the MicroSD. 

The setup I'm currently trying to achieve is having Raspberry Pi OS along with Debian and two ProjectSpaces on the MicroSD slot and also flash Pinn to a USB which will have RetroPie, Zynthian, and four ProjectSpaces.

procount, 2023-07-28:
Use RPI-Imager to flash PINN to your USB drive.
Remove your microSD card altogether and plug your USB drive into your Pi.
Boot PINN on your USB drive and install your OSes to it.

If you can't boot PINN from USB, you may need to adjust your eeprom settings. Plenty of instructions on the Raspberry forums and documentation pages on how to do that.

I'm not aware of Zynthian. I haven't converted it for PINN so you may have to convert it yourself. Generic instructions are available on my wiki and in the README_PINN.md file.

V01D-ST3R, 2023-07-29:
Thanks for the assistance so far. I sort of have it working now. Since the MicroSD has Pinn, it will take priority and the USB will be ignored. The only option I see for my setup is leaving the MicroSD slot empty and using a MicroSD to USB adapter and having to switch between both USB adapters for using the different Pinn installations (reason I'm doing this is because I have to remove fan from gpio, take apart my Pi4 case and disconnect the dsi screen to access the MicroSD slot). 

Also is there a way for Pinn to autobackup/recover the cmdline.txt for installed OS's? I have ZynthianOS booting perfectly with Pinn but when I make a change in the zynthian.local(the local webpage to change settings like display,audio,etc..) the ZynthianOS cmdline.txt "root=PARTUUID=a19bfd3c-09" gets reset to "root=mmcblkxxx" and I have to run "Fix>File System Check" to copy the second "PARTUUID=xxxxxxx" and paste it in "root= from cmdline.txt". 
It would be great if there was a button to quickly restore the default "cmdline.txt" that was set upon OS installation.

procount, 2023-07-29:
Why dont you change the boot priority order to USB first, followed by SD card. Then the SD card will boot with no USB drive, but the USB drive will boot in preference when fitted.

Have you tried the Fix> rerun partition script option?

V01D-ST3R, 2023-07-29:
That's the main issue I encountered. Boot priority is already set to boot from USB first, but it seems like USB is routing Pinn to the MicroSD. I think this may have something to do with "root=/dev/ram0" from the "recovery.cmdline" but when I remove the MicroSD Pinn boots normally from the USB which is strange.

I tried the option to re-run partition_setup but I assume this doesn't work for custom OS's as it shows
 "Checking for partition_setup.sh scripts... ZynthianOS does not support re-running the setup script. ZynthianOS partitions_setup.sh script error".

V01D-ST3R, 2023-07-30:
I also just realized that I can't replace the Project Spaces with any OS I select, I'm trying to install OSMC_Pi4 to project space 1 but after pressing replace it shows the error "Cannot Reinstall/Replace OSMC_Pi4. Partition not big enough for new image." The reinstall button also shows the same error. I noticed there's red text at the bottom which shows "Needed 219MB" but there should be enough space. The MicroSD /W USB adapter is 128GB and has installed RaspiOs, Debian64pi4, and 2 Project Spaces. But since storage is split equally between all 4 OS's there should be about 32GB for each. What am I doing wrong with the replace function?

procount, 2023-07-30:
You can only replace an OS with a similar OS that has the same number of partitions and whise minimum size is less than that already installed since PINN's replace function does not alter the partition layout.

A project space is an empty OS that has 2 partitions. A fixed boot partition and a variable root partition. They were designed for Raspbian. Probably the boot partition is too small for OSMC. 

You could download a project space to usb, then modify its partitions.json file to make one with a larger boot partition, or even with more partitions. Rename it to distinguish it from the others.

Over time the boot partition requirements have expanded. I will look at resuzing them.

procount, 2023-07-30:
Is ZynthianIS available somewhere as a PINN installable version?

V01D-ST3R, 2023-07-30:
Not that I'm aware of, I followed the instructions from https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN and I also uploaded the OS folder to here https://github.com/V01D-ST3R/Zynthian-4-Pinn there are 3 different links(Sync, Mediafire, and Google Drive).

procount, 2023-07-30:
Well done. I'll take a look.

V01D-ST3R, 2023-07-30:
sounds good, the only issue I've seen with ZynthianOS is the one I previously mentioned. root=PARTUUID=XxxXXxx-09" gets reset to "root=mmcblkxxx" after changing settings in the webconfig page(Zynthian.local).

procount, 2023-07-30:
That will be a Zynthian issue. It doesnt expect to be relocated.

V01D-ST3R, 2023-07-30:
Yes I noticed when I first installed it to a USB drive. It's not a big issue as it will only happen during the initial setup or after updates.