procount/pinn #746: Unable to backup armhf partition

RanchoHam, 2023-08-14:
Installed armhf and data partitions from PINN.  Activated armhf partition through normal Raspberry Pi means.  When finally got OS up, I updated the OS, and rebooted.  Upon reboot, I attempted to backup the freshly made armhf partition.  After it showed the filename dialog, the PINN app froze with the maintenance dialog greyed out and the cursor mid-screen approximately where the OK button in the filename dialog had been.

procount, 2023-08-15:
Do you mean the armhf version of the Raspberry Pi OS?
Which model of Pi?
How was your USB drive formatted?
Was this repeatable? Or would it work if you tried
It again?
If it freezes again are you able to access the recovery  shell from the keyboard (ctrl-alt-f2) or via ssh? If so capturing the log and output of dmesg may be useful.

RanchoHam, 2023-08-15:
OK,

Yes, the armhf verson of the Raspberry Pi OS

Model 3B (arm7l)

USB drive is a 64GB Dogfish SSD.  To be truthful, I don't remember how.  Possibly as delivered.  It has been working for other OS backups.

I has been repeatable over the last couple of days that I have tried it.  It is also repeatable across two different Pi3b machines.

I have not yet tried the ctl-alt-f2 approach.  Not sure I have ever set up ssh to PINN.

Will try the ctl-alt-f2 within the next hour.

Cheers,

Rich

procount, 2023-08-15:
The format type of your usb//ssd would be sufficient (e.g fat32, ext4, ntfs etc)

RanchoHam, 2023-08-15:
fat32
ctl-alt-f2 does not respond
"frozen" machine does respond to a ping from another machine on the network
ssh connection refused
I installed dietPi (32bit) and was able to backup that installation.
As I recall you and I are in different hemispheres, so I will wish you a pleasant good day as I go to bed.  Please leave me a list of any more information or tests you need done, and I will do them tomorrow.
Cheers.

RanchoHam, 2023-08-15:
More info:  it also freezes when backing up a raspios arm64 lite installation.

procount, 2023-10-15:
USB disconnects mid-backup. Under investigation.