procount/pinn #751: Backup feature not working

TheUnkn0wnHack3r, 2023-09-17:
Hi. I can't get the backup feature to work. I am using a Raspberry Pi 4. I have a 64GB USB stick with 2 partitions (one labeled 32GB NTFS, one 32GB FAT32, labeled "BACKUP". It contains a os folder and nothing else). Why won't it work?

procount, 2023-09-18:
There are several constraints to using the backup feature. 
Do you mean the backup button is greyed out so can't be pressed?
Perhaps it is because your 64GB USB backup drive has 2 partitions on it, as PINN only looks at the first partition.
If your 32GB FAT32 BACKUP partition is partition #2, then it probably won't be detected.
Try swapping the partitions over.

TheUnkn0wnHack3r, 2023-09-19:
Unfortunatly, after trying this it still didn't reconize it.

procount, 2023-09-19:
Which OSes are you trying to backup?
Can you post some logs? You can look at the troubleshooting section on the wiki for instructions on how to do that.

TheUnkn0wnHack3r, 2023-09-20:
I'm trying to back up "Raspberry Pi OS Lite (64-bit)". However, I have made a major modification that I think I should mentioned. I upgraded the base system (Debian 11 Bullseye) to Debian 12 Bookworm. Should I try a spare 16GB USB that I have lying around (I am using a 32GB micro-sd card for my Pi)
P.S. I will try to post some logs tomorrow

TheUnkn0wnHack3r, 2023-09-22:
So I tried using a diffrent USB with 16GB and only one partition, the drive was detected! However, will a 16GB USB be able to hold a few months of backups (done monthly) for the Pi with a 32GB micro SD card?

procount, 2023-09-22:
Use `df -h`  and add together the used space of the partitions belonging to the OS you are backing up. As a rule of thumb, if you divide this by 3 it will give you an idea of how much space a backup will take up. It is only rough though.

TheUnkn0wnHack3r, 2023-09-22:
Thanks for the quick reply! Could you explain exactly how the backups work? I'm curious if it's incremental or not.

procount, 2023-09-22:
PINN creates a backup in the same format as a set of installation files for an OS. In this way, the backup can be restored by installing or replacing that backup like a standard PINN os installation. So essentially it is a set of tar files (one per partition) that are compressed, and a set of meta-files to tell PINN how to install them. In this way, no unused space is backed up and the backup can be restored to a different sized partition (provided it is big enough). Please test all your backups.

TheUnkn0wnHack3r, 2023-09-23:
When I try to backup, it fails. I would share the logs, but I don't have a way of extracting them, since the logs are under /tmp. And the default browser doesn't load GitHub properly.

procount, 2023-09-23:
I have seen this problem and I'm looking into it. For some reason the USB drive disconnects mid-backup, causing the backup to fail.