procount/pinn #752: Support for Debian 12 Bookworm?

osnabrugge, 2023-09-22:
Any reason why there is no support for Debian 12 Bookworm?  

They have [official tested Raspberry PI images here](https://raspi.debian.net/tested-images/).

Sadly the [Kubernetes community driven template only supports these images](https://github.com/onedr0p/flux-cluster-template#debian-for-raspi4) and I wanted to find a way to be able to remotely reimage if needed through my PIKVM & TESMART KVM as I won't always have physical access to reimage the SSD drives attached to my master nodes (Pi4b 4g), hence coming across your project which is a creative solution.

I saw your wiki and documentation on building a custom OS, but that's not ideal.   How do I go about taking the actual installer and converting that into PINNs format or is this something only you and the others involved in this project do?

Side note: the PIKVM has local mass storage which you can leverage to connect as a virtual CDROM or Flash Drive across USB.  This of course works fine and it's how I reinstall the OS on AMD64 based machines, but I am curious if it could be leveraged for reimaging via PINN.  Based on reading your repo, it seems possible because the requirements are to attach a USB drive with a partition formatted as EXT4, NTFS or FAT32 with an OS folder at the root with the images.  I could just package that up into an ISO or IMG file and store that on the PIKVM which then can be dynamically attached to whichever PI the TESMART is actively connected to for HDMI & USB.

TheUnkn0wnHack3r, 2023-09-23:
What I did to (kinda) install Debian 12 Boomworm was install Raspberry Pi OS Lite (64-bit), and then point the repos to bookworm, and then "sudo apt dist-upgrade"
Note: This is a summary

osnabrugge, 2023-09-23:
Thank you and that's a creative and Interesting approach.  Not sure if that will pass the test of the community, per the explicit messaging:

> Download the latest stable release of Debian from [here](https://raspi.debian.net/tested-images). Do not use Raspbian or DietPi or any other flavor Linux OS.

A few of us have tried and run into various issues.  I'm honestly close to ditching a mixed architecture and just sticking with AMD64, it's been far more than a headache than it's worth.  Running into issues further down the line with plugins/apps having no support for ARM64 or mixed architecture, but I am stubborn :)

I am also curious if doing a dist-upgrade will solve another issue.  The vendor that I bought my [PI rack from appears to do very little in active development](https://github.com/UCTRONICS/SKU_RM0004).  I've struggled getting their LCD and power button working on every flavor except PI OS Lite (64-bit).  Even leveraging (and modifying) an [ansible script](https://github.com/alexb7217/Pi-Rack-Pro_SKU_RM0004-Install) I discovered didn't help.  Will give a shot and see, but I am close to just re-purposing these PIs / selling them.

TheUnkn0wnHack3r, 2023-09-23:
As far as I know, Raspberry Pi OS is Debian (specifically Debian 11 Bullseye at the time of writing) with some rpi-specific stuff installed. I haven't experienced any issues when I pointed the repos to bookworm. I used the Lite version of Raspberry Pi OS because I wanted to use XFCE and because the UI mods dont work. The UI still works, but it looks like default Debian.

lurch, 2023-09-25:
> As far as I know, Raspberry Pi OS is Debian (specifically Debian 11 Bullseye at the time of writing) with some rpi-specific stuff installed.

Raspberry Pi OS based on Debian 12 Bookworm is currently in closed beta-testing :slightly_smiling_face:  No official ETA yet...

osnabrugge, 2023-10-02:
I have decided to port it myself.  I have successfully installed it locally on several PIs and no issues thus far.  I have created the [following PR](https://github.com/procount/pinn-os/pull/18) in hopes that it can be available in the os_testing section for the broader community.

Until the PR is approved, you can [manually download the files](https://sourceforge.net/projects/pinn-os-files/files/v1.0.0-beta1/) and place them within an os\debian12_arm64 folder at the root of a USD/SD Card formatted as ext4 or NTFS.

procount, 2023-10-02:
Well done!
And thanks for doing this.
I'll take a look for you as soon as I can.

procount, 2023-10-05:
I installed it and it boots.
Will include it after some more checking.

osnabrugge, 2023-11-26:
@procount just following up on this, is this something we can merge at least in testing category?  I realize that Raspian's latest release is now bookworm, but they really are not the same as Debian minimal is much leaner.  If there's still some gaps, let me know if there's anything I can help with.

procount, 2023-11-26:
Sorry, I have been consumed with getting PINN working on the Pi5. Let me see what I can do....