procount/pinn #753: Choose OS installation order

GreyzardBigME, 2023-09-24:
when i try to install Android TV 11 and Raspberry Pi OS the installer tells me that the first one has limited USB support and must installed on /dev/sda1, after the installation it contiously reboot without actually booting into Android while RPi os boots normally; if i check the order of the partitions Android is not the first one in the SSD, how can i edit the installation order?

procount, 2023-09-24:
That looks like a subtle mistake. It just means /dev/sda as opposed to using a PARTUUID reference. The partition order will be sorted. But it does mean your SSD must be /dev/sda and not /dev/sdb, which could happen if you have more than one USB device plugged in. (Note: PINN uses partitions 1,2 &5, so the first OS will start at partition 6 anyway).

To explain further - 
Storage devices are normally referenced in the /dev/ folder
The SD card is always referenced as /dev/mmcblk0. There is no other choice.
The first USB device is referenced as /dev/sda, the second as /dev/sdb etc. but the order in which the USB devices get detected is not always consistent. Therefore, in order to have a consistent reference we use the PARTUUID reference for USB devices where we can. However, not all OSes support this, so for those that don't we have to fall back to using /dev/sda, but with the constraint that only 1 USB drive should be fitted to guarantee correct operation.

GreyzardBigME, 2023-09-24:
how can i solve my problem? i remember older versions of PINN installing first Android TV 11 and then RpiOS, with both being able to boot correctly, while the latest version install RpiOS first and it's the only one able to boot now

procount, 2023-09-24:
Have you tried installing Android TV 11 directly to your SSD without PINN to check it works normally?

GreyzardBigME, 2023-09-24:
makes sense, i will try tomorrow but again, the exact same version of Android TV 11 worked perfectly with PINN before, being installed before RpiOS (and/or any other unknown factor)

procount, 2023-09-24:
I'm just trying to narrow down the problem. 
What's different since the last time it worked for you?
It could be the SSD (some USB/SSD adaptors are better than others)
The SSD may not be compatible with Android TV 11
It could be due to a different version of PINN.

I will try installing it myself to test it, but it may not be immediately. Which model of Pi are you using?

GreyzardBigME, 2023-09-24:
Same SSD and same box with usb cable, everything fine with both OS then suddenly the box is malfunctioning (I guess, the SSD seems to works fine on pc)
Tried a fresh install with the latest PINN release but Android doesn't work, I'm sure I'm doing everything the same but now RpiOS is installed before Android and only the first one works, before changing the box (sata to usb 3.0 adapter) even RpiOS just installed gives me problems, now only Android

GreyzardBigME, 2023-09-25:
Just tried Android TV 11 alone and without PINN, works fine just commenting the "dtoverlay=android-sdcard" line and de-commenting "dtoverlay=android-usb".
now I need to fix the installation order and, when I boot Android TV, I want the pi4 to completely ignore the first HDMI port and use only the second with the 28'' HD-Ready TV

GreyzardBigME, 2023-09-28:
i solved the installation order problem with this site:
https://pinn.mjh.nz/
now there is one real problem and a more "cosmetic one:
- as mentioned before, if i boot Android TV 11 with both screens plugged, on the 22'' fullHD pc monitor everything is fine, on the 28'' HD Ready tv plugged in port 2 i dont see anything and the green led on the pi is always flashing (even if the monitor is not powered), if i unplug the monitor i can see the image on the tv; RpiOS doesn't have any problem with both screens and they are set on 1080p and 1080i respectively.
- i would like RpiOS to do not automatically mount (or even show at all) specific partitions (the group used by PINN and Android TV 11 on the SSD, and the entire MicroSD with RpiOS Lite+OpenMediaVault)

lurch, 2023-10-09:
Problem 1 sounds like a problem with Android TV, not with PINN?

Problem 2 sounds like a "problem" with Raspberry Pi OS, not with PINN?

procount, 2023-10-15:
Problem 2 - Open File manager, Edit - Preferences - Volume Management. 
Untick "Mount mountable volumes automatically on program start-up"