procount/pinn #754: None of my installed OSs are bootable

RanchoHam, 2023-09-25:
Suddenly one day my RPi4 based system with a 1TB USB drive shutdown.  Upon re-powering the system none of my OS partitions would boot.
I tried installing a new OS into one of my blank Project Spaces, but that failed with "not enough space" error.  I did manage to install a very lite version; but while it boots, I can't login.
I plugged to drive into a working raspios machine, and all partitions appear to be intact.
Any suggestions?

procount, 2023-09-25:
My first suggestion is to use your Raspios system to backup anything from your 1TB drive that is important.
Prior to it failing, did you install any raspios based OS?
From PINN's command shell (Ctrl-alt-f2 & login with root/raspberry), please post the output of `lsblk -o NAME, PARTUUID`
and post the output of installed_os.json on the /settings partition (/dev/sda3)

RanchoHam, 2023-10-05:
OK, sorry about the delay; I have been on travel.

This has happened on two of my machines:
1. on this one (which has one bootable OS) all of my partitions were either Raspberry PI OSs, data space or uncommitted Program Spaces.
2. the other machine (with no bootable OSs) has buster 32bit and 64bit, bullseye 32bit and 64bit, Ubuntu, gentoo64, and a data space.

I note that a plain lsblk does not show /dev/sda3 or /dev/sda4 partitions.  There is a **installed_os.json** file on /dev/sda5.  

BTW for anyone else trying to follow these steps there is no space in the column list for lsblk -o, _e.g._ use **lsblk -o name,partuuid**.

Here is the output of **lsblk -o name,partuuid > lsblk.txt** from the machine where I can get one partition to boot:
[lsblk.txt](https://github.com/procount/pinn/files/12819172/lsblk.txt)

Here is the **installed_os.json** file from that same machine (renamed with a .txt extension):
[installed_os.json.txt](https://github.com/procount/pinn/files/12819250/installed_os.json.txt)

I'll work on getting the info from the completely failed machine.

procount, 2023-10-05:
No need. I see the problem.

RanchoHam, 2023-10-05:
Alright!  I have both systems back on-line with their original content without wiping, re-installing, or restoring from backup!
Huge thanks to @procount, @lurch, and @billsidea for isolating the issue and identifying a fix!

closed 2023-10-05 by RanchoHam