procount/pinn #755: Please help me to setting up generic touchscreen

jandes62, 2023-10-04:
Hi,
I connected an SPI display with touchscreen to my raspberry, it's an ILI9341 2.8" with XPT2046 touch controller;
it works fine in the installed OS with this options:

`dtparam=spi=on
dtoverlay=fbtft,spi0-0,ili9341,bgr,width=240,height=320,reset_pin=23,dc_pin=24,led_pin=25
dtparam=rotate=270
dtoverlay=ads7846,penirq=22,speed=50000,penirq_pull=2,xohms=150`

I would like to make it work in pinn to choose the boot OS
I added the same options in the config.txt in RECOVERY folder but it doesn't work

Is there anyone who could help me? I'm a complete newbie
I apologize for my bad english

thanks

procount, 2023-10-05:
PINN is very small and only contains a minimal set of common device drivers. Drivers for your display and touchscreen would need to be added.

jandes62, 2023-10-06:
Thanks for your reply
Are HD44780 LCD displays supported?
A good alternative for me could be to send the two options to the LCD and select via GPIO push button. Ideally before starting PINN and, based on your selection edit the autoboot.txt file like

`if GPIOn=true
     echo 'boot_partition=6' > autoboot.txt`
else
     echo 'boot_partition=8' > autoboot.txt`
or
`[GPIOn=0]
boot_partition=6
[ALL]
boot_partition=8`
is it possible to do ?

jandes62, 2023-10-06:
Thanks for your reply
Are HD44780 LCD displays supported?
A good alternative for me could be to send a OS menu options to the LCD and select via GPIO push button. Ideally before starting PINN and, based on your selection edit the autoboot.txt file like

`if GPIOn=true echo "boot_partition=6" > autoboot.txt
else echo "boot_partition=8" > autoboot.txt`

or an autoboot.txt like this

`[GPIOn=1]
boot_partition=6
[ALL]
boot_partition=8`

is it possible to do ?

procount, 2023-10-06:
PINN only supports the RPT 7" LCD touchscreen and the Pimoroni Hyperpixel 3.5" & 4" DPI displays in addition to standard HDMI and Composite displays.
There is a pinn_init.sh script that runs before PINN to setup displays or detect / switch between multiple displays. This is where displays may be configured, but they would still require device drivers to be included in the kernel for them to work.
You cannot use such scripts to modify autoboot.txt, because that is read very early in the boot process before any scripts are run. Therefore you would have to reboot after changing it.
Conditional code could be used in autoboot.txt to load different OSes based on a GPIO value. In this case you would have no display, just activate a particular GPIO when powering up to select a particular OS.

lurch, 2023-10-06:
Also, IIRC if you're using `autoboot.txt` that then bypasses PINN, so the only way to "return to PINN" is to delete `autoboot.txt` ?

procount, 2023-10-06:
Correct. Good to point it out.

jandes62, 2023-10-09:
> PINN is very small and only contains a minimal set of common device drivers. Drivers for your display and touchscreen would need to be added.

it is possible to install the necessary drivers from within pinn console without recompiling the kernel or rebuild PINN?

procount, 2023-10-09:
I fixed the links for you.
I'm not sure about the HD44780 driver. 
PINN won't display on it. 
You'd probably have to control it yourself from a shell script.

lurch, 2023-10-09:
> PINN won't display on it.

What??? You need to brush up on your pixel-art skills @procount :rofl:

jandes62, 2023-10-09:
> Ho corretto i link per te. Non sono sicuro del driver HD44780. PINN non verrà visualizzato su di esso. Probabilmente dovresti controllarlo da solo da uno script di shell.

sure,
I imagine something like sending a name of OS to the LCD display via lcdproc and navigating the list of available OSes  via UP/DOWN push buttons (even better with rotary encoder :winking_face: )

procount, 2023-10-09:
> What??? You need to brush up on your pixel-art skills @procount 🤣

Lol 🤣 It'd be like painting the inside of the Royal Albert Hall through the letterbox!
Brings back memories of User Defined Graphics on the ZX81.

> I imagine something like sending a name of OS to the LCD display via lcdproc and navigating the list of available OSes via UP/DOWN push buttons (even better with rotary encoder :winking_face: )

Could be possible. You'd have to do the processing in pinn_init.sh: process the installed_os.json file, deal with any GPIO inputs and boot into the required partition.
I had imagined including something similar along with some other architectural changes to PINN, but given the (lack of) spare time I have, it might take a good while.

jandes62, 2023-10-09:
I imagine it because I am not able to realize it 😂

lurch, 2023-10-09:
:thinking: 
Just out of curiosity, if you're using a Pi where the only means of interaction is a 16x2 LCD and a few push-buttons, what are you doing that means that you still need to be able to switch OSes using PINN?

jandes62, 2023-10-09:
I want to make a multimedia streamer, with Volumio for music and OSMC for movies etc... the raspberry is also connected to the TV but this is only activated with Kodi and I want to be able to choose OS without having to turn it on when I use Volumio

jandes62, 2023-10-10:
both Volumio and OSMC are operating systems themselves, with a minimal footprint, you can't install one inside the other, I know there is a kodi plugin for the old version of Volumio, but it's not stable.

lurch, 2023-10-14:
@jandes62 Perhaps you could try the approach described in #756 ?

jandes62, 2023-11-01:
@lurch I know this method, you have to choose the OS before power on the rpi; if I have no other alternatives I will have to use this but I would prefer to be able to choose the OS after switching on, possibly via the touch screen.
Thanks