procount/pinn #756: tmp/mnt/pinn_init.sh not found

axalppro, 2023-10-10:
Hello, 
I am trying to recreate this : [OS selection with jumper](https://forum.dexterindustries.com/t/how-to-jumper-select-configuration-boot-options-for-the-raspberry-pi/8013/17)

First, thanks for all the hard work and support for this kind of project.

I have an SD card running PINN and two OS installed. I want to change the one that boots using a jumper similar to what has been done in the post above. I modified the config.txt and added the pinn_init.sh accordingly. When I try to boot I get the rainbow box and then a line that says : 
`sh: /tmp/mnt/pinn_init.sh: not found`

So I'm guessing it detects the pinn_init.sh script tries to mount it but fails and then can't run it.

Any idea where this error could be coming from.

PS: I modified the pinn_init.sh for my application so maybe I have an error in it but it didn't work better with the original pinn_init.sh povided in the article by Jim.

Here is my pinn_init.sh file : 
```
#!/bin/sh
# This function will export a GPIO number as an INPUT so it can be used
gpio_export()
{
	#arg1 = gpio number
	echo "$1" >/sys/class/gpio/export
	echo "in" >/sys/class/gpio/gpio$1/direction 
}

# This function will free up the GPIO for use elsewhere
gpio_unexport()
{
	echo "$1" >/sys/class/gpio/unexport
}

# This gets the value of the GPIO pin
gpio_value()
{
	#arg1 = gpio number
	cat /sys/class/gpio/gpio$1/value
}

# Initialise the partition number to 4
partition=4	# Start with a number that, when
		# incremented by two will hit each boot partition in turn

# Identify which OS we want to load by the first GPIO number that is set
# If more than one is set, the first one detected is selected.

n=27
#  Valid bootable partitions are the even numbered partitions beginning with partition 6
#  We start with $partition = 4 because we pre-increment
partition=`expr $partition + 2`		#  Increment partition number to next untested partition
gpio_export $n
value=$(gpio_value $n)
echo "exported pin $n returned a value of $value"

#  Test the returned value from each GPIO pin.
#  A selected pin will return "0" and will trigger the boot process
if [[ "$value" == "0" ]]; then
	echo "Jumper ON - rebootp $partition"	#  Display what's going to happen
	rebootp $partition		#  Force-boot the selected partition
	break				#  Never reach here unless testing
else 	
	echo "Jumper OFF - rebootp $partition"	#  Display what's going to happen
	partition=`expr $partition + 2`		#  Increment partition number to next untested partition
	rebootp $partition		#  Force-boot the selected partition
	break
fi
gpio_unexport $n
	

echo "No O/S selected - booting PINN"
```

And my config.txt : 
```
gpio=27=ip,pu

gpu_mem=16
start_file=recovery.elf
fixup_file=fixup_rc.dat

[pi4]
start_file=recover4.elf
fixup_file=fixup4rc.dat
max_framebuffers=2

enable_uart=1
```

axalppro, 2023-10-11:
Here's debug.txt : 
```
Keyboard found: "/dev/input/event1" 
3.8.7 
Board revision is  13644052 
Found recovery.rfs at "mmcblk0p1" 
PINN drive: "/dev/mmcblk0" 
mount: /dev/mmcblk0 is already mounted or /mnt busy
Simulate OK 
CecListener thread started 
Looking for  "/mnt/cec_keys.json" 
CecListener done initializing 
Loading Mappings mappings from  ":/cec_keys.json" 
Shift detected 
New =  mainwindow ,  Main Menu 
setWindow  "mainwindow" 
setMenu  "Main Menu" 
Starting dbus 
Starting dhcpcd 
Default language is  "en" 
Default keyboard layout is  "gb" 
Default display mode is  "0" 
Number of bootables =  2 
New drive selected: "mmcblk0" 
Network up in 5.941 seconds 
current date is  QDate("Thu Jan 1 1970") 
Requesting current time 
Downloading "http://downloads.sourceforge.net/project/pinn/BUILD-DATA" to "BUILD|/tmp/BUILD-DATA" 
Downloading "http://downloads.sourceforge.net/project/pinn/README.md" to "README|/tmp/README.md" 
Downloading "http://raw.githubusercontent.com/procount/pinn-os/master/os/overrides.json" to "GROUP|/tmp/overrides.json" 
downloadRepoList:  "http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json" 
Time set to:  QDateTime("Wed Oct 11 06:27:47 2023") 
Redirection - Re-trying download from "https://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json" 
downloadRepoList:  "https://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json" 
Downloading "https://raw.githubusercontent.com/procount/pinn-os/master/os/overrides.json" to "GROUP|/tmp/overrides.json" 
"GROUP" 
Downloaded  "GROUP" : "/tmp/overrides.json" 
Number of bootables =  2 
Downloading "http://kumisystems.dl.sourceforge.net/project/pinn/BUILD-DATA" to "BUILD|/tmp/BUILD-DATA" 
Downloading "http://netcologne.dl.sourceforge.net/project/pinn/README.md" to "README|/tmp/README.md" 
Number of bootables =  2 
checking size of file:  "http://downloads.sourceforge.net/project/pinn/os_previous/recalbox7-rpi4/RECALBOX.tar.xz" 
Number of bootables =  2 
checking size of file:  "http://downloads.sourceforge.net/project/pinn/os/kali64/boot_kali.tar.xz" 
checking size of file:  "http://downloads.sourceforge.net/project/pinn/os/kali64/root_kali.tar.xz" 
"BUILD" 
Downloaded  "BUILD" : "/tmp/BUILD-DATA" 
"README" 
Downloaded  "README" : "/tmp/README.md" 
BUILD_IGNORE... 
BUILD_CURRENT... 
Version:  "PINN Version: v3.8.7" , "v3.8.7" 
BUILD_NEW... 
Version:  "PINN Version: v3.8.7" , "v3.8.7" 
Number of bootables =  2 
checking size of file:  "http://download.osmc.tv/installers/noobs/boot-rbp4.tar.xz" 
checking size of file:  "http://download.osmc.tv/installers/noobs/root-rbp4.tar.xz" 
checking size of file:  "http://altushost-swe.dl.sourceforge.net/project/pinn/os_previous/recalbox7-rpi4/RECALBOX.tar.xz" 
libpng warning: Unknown iTXt compression type or method
libpng warning: Unknown iTXt compression type or method
Number of bootables =  2 
checking size of file:  "http://nav.dl.sourceforge.net/project/pinn/os/kali64/boot_kali.tar.xz" 
checking size of file:  "http://deac-riga.dl.sourceforge.net/project/pinn/os/kali64/root_kali.tar.xz" 
Length: 753328276 files left:  4   "recalbox7-rpi4" 
checking size of file:  "http://ftp.halifax.rwth-aachen.de/osmc/osmc/download/installers/noobs/root-rbp4.tar.xz" 
checking size of file:  "http://mirrors.tuna.tsinghua.edu.cn/osmc/osmc/download/installers/noobs/boot-rbp4.tar.xz" 
Number of bootables =  2 
checking size of file:  "http://rpi.thinlinx.com/NOOBS/TLX_RPi/boot.xz" 
checking size of file:  "http://rpi.thinlinx.com/NOOBS/TLX_RPi/tfm.xz" 
checking size of file:  "http://rpi.thinlinx.com/NOOBS/TLX_RPi/root.xz" 
[... 294 lines cut ...]
[    3.574331] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.451937] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e
[    5.449713] uart-pl011 fe201000.serial: no DMA platform data
[    6.532035] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[    7.235775] input: uinput-sample as /devices/virtual/input/input2
[    7.647695] EXT4-fs (mmcblk0p5): mounted filesystem with ordered data mode. Opts: (null)
[    8.968141] EXT4-fs (mmcblk0p5): mounted filesystem with ordered data mode. Opts: (null)
[    9.006416] random: dbus-uuidgen: uninitialized urandom read (12 bytes read)
[    9.006443] random: dbus-uuidgen: uninitialized urandom read (8 bytes read)
[    9.019334] random: dbus-daemon: uninitialized urandom read (12 bytes read)
[    9.210051] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    9.339584] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    9.339789] bcmgenet fd580000.ethernet eth0: Link is Down
[   11.445936] bcmgenet fd580000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   11.454440] urandom_read: 2 callbacks suppressed
[   11.454445] random: dhcpcd: uninitialized urandom read (120 bytes read)
[   12.025730] random: dhcpcd: uninitialized urandom read (120 bytes read)
[   12.025869] random: dhcpcd: uninitialized urandom read (120 bytes read)
[   15.081560] urandom_read: 1 callbacks suppressed
[   15.081566] random: dhcpcd: uninitialized urandom read (120 bytes read)
[   15.149710] random: crng init done
[   32.485701] cam-dummy-reg: disabling
[   32.485717] cam1-reg: disabling
[  174.863384] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
```

procount, 2023-10-11:
There's nothing obvious in there.
I will try to replicate it.

axalppro, 2023-10-12:
Thanks !

Also I tried your savelog script. I get this error.

```
root@recovery:~# /mnt/savelog
find: /tmp/media: No such file or directory
root@recovery:~#
```

And just to make sure I understand the process right, pinn_init.sh is run before Pinn has booted meaning that using the rebootp command in the pinn_init script should prevent the Pinn GUI etc. to appear ?

procount, 2023-10-12:
It should still have produced a /mnt/log file. It hust means you didn't have any USB drives connected.

Yes, pinn_init.sh is run from init, but as it can't be executed for some reason, init continues to boot PINN.

axalppro, 2023-10-12:
Okay, I tried a new clean install on another hardware using a Compute Module 4, as I have to connect via VNC, I don't see the error but Pinn boots so I think the script isn't executed. I check the debug and dmesg files nothing seems suspicious.

Do you have any idea where I could look into next ?

procount, 2023-10-12:
I think you might have DOS line endings for pinn_init.sh instead of UNIX line endings.
Please check by going to PINN's command shell and type `nano /mnt/pinn_init.sh`
If you see a `^M` at the end of every line, you have DOS line endings and the `^M`s must be removed.

In linux, you can use the dos2unix tool to remove them.
In Windows you can use notepad++ to change the line endings.

axalppro, 2023-10-13:
That was it, now it works.  So awesome to change OS with a jumper. Thanks for the support !

closed 2023-10-13 by axalppro