procount/pinn #758: Raspberry pi 3 Touchscreen

tinytempmouse, 2023-11-03:
The touch screen works fine but screen rotation doesn't not want to work I've use latest version of PINN and have tried lcd_rotate=2 in config.txt, and display_rotate=2 but still it's upside down.

procount, 2023-11-03:
Are you referring to the official Raspberry Pi 7" DSI touchscreen?

procount, 2023-11-03:
Using the official touchscreen with a RPi3, I flipped the screen upside down using `lcd_rotate=2` command.
`display_lcd_rotate=2` only flips the display but not the touch.
Make sure you put it ABOVE the [pi4] section, otherwise it won't work unless you are using a Pi4 :wink:

CONFIG.TXT
```
gpu_mem=16
start_file=recovery.elf
fixup_file=fixup_rc.dat

lcd_rotate=2

[pi4]
start_file=recover4.elf
fixup_file=fixup4rc.dat
max_framebuffers=2
```