procount/pinn #759: Raspberry Pi 5 Support

br6, 2023-11-04:
Will you be adding RPI5 Support?

thanks

Sulphy77, 2023-11-22:
Yes please.... I am also eager to know if pinn will be updated to support RPi5

jaca2300, 2023-11-26:
Please,,,

procount, 2023-12-02:
Sorry, github has stopped sending me notifications so I missed many messages - I will have to fix that.
Anyway, I have been working on PI5 for several months. Finally I am seeing light at the end of the tunnel. It will still be a little while yet, but it's coming....

Macrobiotic, 2024-01-03:
Just chasing for an update on Pi5 compatibility. Version 3.8.9 seems to get as far as letting us install OS-es, but when booting into one of them the Pi5 appears to reboot. Any work-around?

Using a micro SD from Samsung (512 Pro Ultimate).

procount, 2024-01-03:
It is essenrial to upgrade your firmware, as detailed in the release post https://forums.raspberrypi.com/viewtopic.php?p=2173713#p2173713

Macrobiotic, 2024-01-03:
Many thanks for the quick reply.
I've shared this info with the discussion thread under this YouTube video: https://www.youtube.com/watch?v=WVCNPCHN4xU

The video is all about trying to get PINN working on a Pi 5.

Macrobiotic, 2024-01-03:
The readme file in the rpi-eeprom-recovery.zip indicates it's intended for Raspberry Pi 4 specifically.
Is it safe to run it on Pi 5?

procount, 2024-01-03:
Yes. they (RPL) probably forgot to update the readme.
It is essential to get multi-boot working. A small regression when they were getting Power Delivery sorted.

Macrobiotic, 2024-01-03:
Got it. I'll give it a try.
Thanks for the quick reply.

Macrobiotic, 2024-01-03:
Yep, that one works. Successful boot into Ubuntu 23.10.
Hope RPL get around to releasing a fixed firmware.

procount, 2024-01-03:
I don't know their timescale, but it will eventually make it into a future `apt-update; apt full-upgrade`

lurch, 2024-01-03:
It looks like it was fixed in the `pieeprom-2023-12-14` release https://github.com/raspberrypi/rpi-eeprom/commit/3f325bd482649e697ddf46c1484a1a7c7e25d6e5
But the most recent "full release" was `rpi-boot-eeprom-recovery-2023-12-06-2712` https://github.com/raspberrypi/rpi-eeprom/releases

So as you say, hopefully it won't be too long for things to get updated in `apt` too.

olly69, 2024-01-10:
Has anyone managed to get PINN working on a pi 5 booting from NVMe?  I'd like a dual boot 64 and 32 bit Rasp os...

Thanks!

Sulphy77, 2024-01-16:
I have a pineberry hat top that should be coming this weekend, so ill give it a try... am sure you already know if this hat, but here is the link just in case :)
https://pineberrypi.com/products/hat-top-2230-2240-for-rpi5
Will give PINN a go ... and report back! Thanks again!

Poopi, 2024-01-18:
I use NVM connected to Rpi5 via PiHat.
I've installed Ubuntu and Pi OS. Pi boots to busybox but it is not finding the root partition. Ubuntu cannot even unpack rootfs. See attached screenshots.
Without PINN I can boot NVM with either Ubuntu or PiOS without any issues.
Any suggestions for further analysis?

  [image]
[image]

ardoviniandrea, 2024-01-18:
I've been testing the PIMORONI NVME base with PINN.

So far:
- raspiOS works perfectly
- lineage20 ATV works perfectly (even the TWRP recovery)
- ubuntu 23, as poopi reported, stops at boots.

I'm happy to do more testing whenever needed, and I'll continue to play around myself and report here if I find something that's not working properly

Poopi, 2024-01-18:
Hmmm. It should not be that different from what I'm doing. So let me start messing around with configs ...

ardoviniandrea, 2024-01-23:
Hello procount

Can you please provide a bit more detail on your last comment?
I have connected my PINN SSD to a NVME reader so I have access to all partitions.

I do not see a SETTING partition and the RECOVERY partition has an empty /os/ folder

I have a lot images installed (and all the Project Spaces for future plans and tests), so it's possible I'm getting confused between folders.
I have attached a screenshot of my partitions.
[image]

procount, 2024-01-23:
@ardoviniandrea - Are you viewing this on an Apple device? I am not familiar with those. As you have all the other partitions visible, and you can boot PINN, then the settings partition must be there. Perhaps Apple devices don't show partitions named "SETTINGS" ? I know that many OSes (e.g. Ubuntu) don't show partitions named "RECOVERY".

From the RPi forums, I have discovered that Ubuntu 23.10 seems to have problems with some USB hubs on the Pi5. May be the same issues are happening with NVME, so this partition_setup script may not make any difference.

Anyway, as a workaround, I suggest you copy the partition_setup.sh script to the RECOVERY partition and boot PINN on that device on your RPi. Press Shift when prompted so that you go into the recovery menu.
From the recovery menu press [CTRL]-[ALT]-F2 to open a terminal and login with `root/raspberry`.
From here, the settings partition should already be mounted as /settings. so type:
```
cd /settings/os/ubuntu2310
mv partition_setup.sh partition_setup.bak
cp /mnt/partition_setup.sh .
```
(Don't miss the dot/period in the last command! Please check the above for spelling/accuracy etc. as I am recounting this from memory!)

Return to PINN by pressing [CTRL]-[ALT]-F1
Go to the maintenance menu.
Tick the box next to Ubuntu to enable the Fixup button.
Select the Fix button
tick the box to re-run the partition_setup.sh and press OK.
Hopefully it will complete successfully, and then you can try and boot Ubuntu. (Probably won't make any difference but worth a try)

procount, 2024-01-23:
An empty /os folder on RECOVERY is to be expected.

procount, 2024-01-23:
@Poopi 
> Any suggestions for further analysis?

Could you post copies of `/settings/installed_os.json` along with copies of `cmdline.txt` from your Ubuntu boot partition and `/etc/fstab/` from your Ubuntu root partition?

Sulphy77, 2024-01-24:
So i got my pineberry hat top, and after MANY attempts, finally managed to get booting from the nvme to stick! I had to re-image the sd card quite a few times... as well as the nvme drive... setting the boot config etc... as the nvme would install raspios just fine...  but if i did a system update from the prompt for first login... when rebooting, it would just drop me right into the busybox cmd screen (which i have no clue what to do with btw!!! n00b) ... re-image PINN... bootup from sd card... redo boot config etc.... last night tho, and this morning... i managed to get it working as expected, multiple reboots later... and still good!
So, long story short, at least in my case, do not run the updater when prompted on first boot! reboot a few times first... then do the update via ssh/terminal! Looking forward to the next version of Pinn!!!

PS: Would be awesome if it could support wubuntu or zorin (i know they arent pi compatible... just wish they were :D )

procount, 2024-01-24:
That's good news!
So what is your setup?
Have you got PINN on the sd card that you've used to install Raspios on your nvme?

Sulphy77, 2024-01-24:
Sure thing, i will install them all... document each one as i go through them... and report back! Gimme a few days to play as and when i can, i mess about with the pi5 in between work!

ardoviniandrea, 2024-01-24:
Some updates here.

I've tested what procount  shared in the previous comment.
The Fix run successfully but as anticipated, didn't solve the issue for Ubuntu that still stops here immediately after the boot.
[image]

I did test most of the other OSes and I can report that all of these works correctly when booting from PINN on NVME:
- RaspOS arm64
- kde2lee64
- AOSP14-rpi5 (although I managed to brick it somehow and required to reinstall it)
- lineage20-rpi5 (with TWRP and successfully installing Gapps)
- lineage20-rpi5-ATV (with TWRP and successfully installing Gapps)
- mxlinux
- LibreELEC_RPi5
- recalbox5 (boots fine, but I haven't try any games for so far, so I cannot report the full funcionality)
- ProjectSpace --> I've initially installed all 8 possible, and already used/swapped a few without any issues.

2 notes I want to report, other than Ubuntu not working:
- HDR, HDR+ and HVEC content in KODI in LineageOS (even ATV) plays with washed out / grey colors, but plays normally in LineageOS.
I looked around a bit and looks like it could be related to codec issue, but haven't been able to identify more than this
- PINN boot related, very minor but I am not able to re-order the OSes in the boot menu using the UP and DOWN buttons. Every time I do it and either boot or turn off the RPI, the order comes back to the initial one the next boot.

Happy to keep testing new thing procount, if you have anything in mind

Poopi, 2024-01-24:
Hah. Glad to see that I'm not the one. I bet that Ubuntu it will work like a charm when you use USB->NVMe dongle.

procount, 2024-01-24:
@ardoviniandrea - Thanks for testing.
On your 2 points:

- I'm not sure how installing via PINN would affect the codecs. I wonder if you get the same effect if you could attach your NVME via USB, or if you install via PINN on an SD card?
- You must choose the BOOT option after re-ordering the OSes to save the new order. Is this only happening on NVME or have you experienced it on other drives?

Sulphy77, 2024-01-26:
Ok, so i started messing about with the rpi5, pinn & multiple OS'd... i couldnt find any solid detail on running multiple OS's... am sure on my older pi's with pinn, there was an option to set partitions up for multiple os's.... but, for now anyways, i have gone down the route of replacing raspios with ubuntu... giving this a whirl now! If you have any "dunny" guide that outlines how to setup multiple os;s.. with a boot menu or such... would be appreciated! Will report back on ubuntu... :)

procount, 2024-01-26:
@Sulphy77 - sorry, I don't understand your question. One of PINN's main purposes is to allow multiple OSes to be installed to a single drive and provide a boot menu to select the OS you want to run on startup. If you have only installed one OS, then you won't get a boot menu because there is no choice to be made.

To install multiple OSes, you just check all the OSes you want (across the various categories, then press the Install button. PINN will allocate the partition space for you. It shows you in the bottom left corner how much space is required and whether your selections will fit. I advise not to go close to the limit because each OS will then have minimal space itself and no room to store any additional data or apps.

Note that whatever OSes you install together at the beginning, determines the layout of the partitions thereafter. PINN does not subsequently change this layout without doing a full install again. You can backup, restore or replace your OSes provided the OS being restored or replaced matches the previous OS's partition layout. You can use ProjectSpaces to reserve space for future OSes that you may want to install, but these are standard 2 partition layouts and therefore not suitable for OSes like Android or Recalbox that require more than 2 partitions.

You can find the user guide here-> https://github.com/procount/pinn/blob/master/README_PINN.md

Sulphy77, 2024-01-26:
Thanks Procount! Yep, that makes sense... like i may have mentioned... not much of a linux guy!!! :D :D :D ... appreciate your patience and support!

Sulphy77, 2024-01-26:
Ubuntu on my rpi5, tried the suggested fix of the partition.sh file... and no joy. I also get the kernel panic :P

procount, 2024-01-26:
Oh well, it was worth a try. 
Ubuntu 23.10 works on the Pi5, on USB, SD card and NVMe
It works with PINN on the pi5 on USB and SD card.
So there must be something odd with the combination of PINN, PI5 and NVMe. 😕

Poopi, 2024-01-26:
Not sure if this helps but this is the output on the serial port during Ubuntu booting:
RPi: BOOTLOADER release VERSION:30cc5f37 DATE: 2024/01/05 TIME: 15:57:40
BOOTMODE: 0x06 partition 6 build-ts BUILD_TIMESTAMP=1704470260 serial ee821091 boardrev d04170 stc 4591535
AON_RESET: 00000003 PM_RSTS 00001034
usb_pd_init status 3
USB_PD CONFIG 0 43
Boot mode: NVME (06) order f41
PCI1 init
PCI1 reset
PCIe scan 0000c0a9:0000540a
VID 0xc0a9 MN CT1000P3PSSD8                          
NVME on
MBR: 0x00002000,  260096 type: 0x0e
MBR: 0x00041800,1953256880 type: 0x05
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
USB-PD: src-cap PDO object1 0x0a0191f4
Current 5000 mA
Voltage 5000 mV
USB-PD: src-cap PDO object2 0x0002d12c
Current 3000 mA
Voltage 9000 mV
USB-PD: src-cap PDO object3 0x0003c0e1
Current 2250 mA
Voltage 12000 mV
USB-PD: src-cap PDO object4 0x0004b0b4
Current 1800 mA
Voltage 15000 mV
Trying partition: 0
type: 16 lba: 8192 ‘mkfs.fat’ ‘  V       ^ ‘ clusters 64887 (4)
rsc 1 fat-sectors 256 root dir cluster 1 sectors 32 entries 512
FAT16 clusters 64887
[nvme] autoboot.txt not found
Select partition rsts 6 C(boot_partition) 0 EEPROM config 0 result 6
Trying partition: 6
EBR: 0x00041800 0x00000800,   65534 0x000107fe,1953189298 signature: aa55
EBR: 0x00051ffe 0x00000002, 1034240 0x0010d000,1952155056 signature: aa55
MBR: 0x00000002, 1034240 type: 0x0c
type: 16 lba: 335872 ‘mkfs.fat’ ‘  V       ^ ‘ clusters 64605 (16)
rsc 1 fat-sectors 256 root dir cluster 1 sectors 32 entries 512
FAT16 clusters 64605
Read config.txt bytes     1612 hnd 0x4c
[nvme] pieeprom.upd not found
usb_max_current_enable default 0 max-current 5000
Read bcm2712-rpi-5-b.dtb bytes    73712 hnd 0x3a
dt-match: compatible: raspberrypi,5-model-b match: brcm,bcm2712
dt-match: compatible: brcm,bcm2712 match: brcm,bcm2712

NOTICE:  BL31: v2.6(release):v2.6-239-g2a9ede0bd
NOTICE:  BL31: Built : 14:26:57, Jun 22 2023

procount, 2024-01-26:
@Poopi - As you can capture debug logs, perhaps you could try adding the following options in your ubuntu configuration? :
In /boot/firmware/config.txt
```
[all]
enable_uart=1
uart_2ndstage=1
dtoverlay=miniuart-bt
```
In /boot/firmware/cmdline.txt, add
`console=serial0,115200 loglevel=7`
and remove `quiet`

This is all on one line.

Poopi, 2024-01-27:
Please find below ( looks like root mounting fails):

> RPi: BOOTSYS release VERSION:30cc5f37 DATE: 2024/01/05 TIME: 15:57:40
BOOTMODE: 0x06 partition 6 build-ts BUILD_TIMESTAMP=1704470260 serial ee821091 boardrev d04170 stc 1605263
AON_RESET: 00000003 PM_RSTS 00001034
RP1_BOOT chip ID: 0x20001927
PM_RSTS: 0x00001034
part 00000006 reset_info 00000000
PMIC reset-event 00000000 rtc 0000001d alarm 00000000 enabled 0
uSD voltage 3.3V
Initialising SDRAM ‘Micron’ 32Gb x2 total-size: 64 Gbit 4267
DDR 4267 1 0 64 152
RP1_BOOT chip ID: 0x20001927

>RP1_BOOT chip ID: 0x20001927
RP1_BOOT: fw size 25968
PCI2 init
PCI2 reset
PCIe scan 00001de4:00000001
RP1_CHIP_INFO 20001927

>RPi: BOOTLOADER release VERSION:30cc5f37 DATE: 2024/01/05 TIME: 15:57:40
BOOTMODE: 0x06 partition 6 build-ts BUILD_TIMESTAMP=1704470260 serial ee821091 boardrev d04170 stc 4591505
AON_RESET: 00000003 PM_RSTS 00001034
usb_pd_init status 3
USB_PD CONFIG 0 43
Boot mode: NVME (06) order f41
PCI1 init
PCI1 reset
PCIe scan 0000c0a9:0000540a
VID 0xc0a9 MN CT1000P3PSSD8                          
NVME on
MBR: 0x00002000,  260096 type: 0x0e
MBR: 0x00041800,1953256880 type: 0x05
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
USB-PD: src-cap PDO object1 0x0a0191f4
Current 5000 mA
Voltage 5000 mV
USB-PD: src-cap PDO object2 0x0002d12c
Current 3000 mA
Voltage 9000 mV
USB-PD: src-cap PDO object3 0x0003c0e1
Current 2250 mA
Voltage 12000 mV
USB-PD: src-cap PDO object4 0x0004b0b4
Current 1800 mA
Voltage 15000 mV
Trying partition: 0
type: 16 lba: 8192 ‘mkfs.fat’ ‘  V       ^ ‘ clusters 64887 (4)
rsc 1 fat-sectors 256 root dir cluster 1 sectors 32 entries 512
FAT16 clusters 64887
[nvme] autoboot.txt not found
Select partition rsts 6 C(boot_partition) 0 EEPROM config 0 result 6
Trying partition: 6
EBR: 0x00041800 0x00000800,   65534 0x000107fe,1953189298 signature: aa55
EBR: 0x00051ffe 0x00000002, 1034240 0x0010d000,1952155056 signature: aa55
MBR: 0x00000002, 1034240 type: 0x0c
type: 16 lba: 335872 ‘mkfs.fat’ ‘  V       ^ ‘ clusters 64605 (16)
rsc 1 fat-sectors 256 root dir cluster 1 sectors 32 entries 512
FAT16 clusters 64605
Read config.txt bytes     1507 hnd 0x244a
[nvme] pieeprom.upd not found
usb_max_current_enable default 0 max-current 5000
Read bcm2712-rpi-5-b.dtb bytes    73712 hnd 0x3a
dt-match: compatible: raspberrypi,5-model-b match: brcm,bcm2712
dt-match: compatible: brcm,bcm2712 match: brcm,bcm2712
Read /config.txt bytes     1507 hnd 0x244a
Read /config.txt bytes     1507 hnd 0x244a
MESS:00:00:04.842825:0: Initial voltage 800000 temp 36730
[... 331 lines cut ...]
[    1.901140]  vfat
[    1.903411] 
[    1.906821] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    1.915116] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 6.5.0-1005-raspi #7-Ubuntu
[    1.922538] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[    1.928388] Call trace:
[    1.930834]  dump_backtrace+0x9c/0x128
[    1.934594]  show_stack+0x20/0x38
[    1.937914]  dump_stack_lvl+0xbc/0x120
[    1.941671]  dump_stack+0x18/0x28
[    1.944991]  panic+0x370/0x400
[    1.948050]  mount_root_generic+0x19c/0x328
[    1.952245]  mount_root+0xa4/0x118
[    1.955652]  prepare_namespace+0x74/0x2d0
[    1.959669]  kernel_init_freeable+0x258/0x2d0
[    1.964037]  kernel_init+0x38/0x218
[    1.967532]  ret_from_fork+0x10/0x20
[    1.971115] SMP: stopping secondary CPUs
[    1.975047] Kernel Offset: 0x200004000000 from 0xffff800080000000
[    1.981159] PHYS_OFFSET: 0x0
[    1.984042] CPU features: 0x00000000,b8014021,8001720b
[    1.989194] Memory Limit: none
[    1.992252] —[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0) ]—

There is also one more thing. In my config the RaspberryOS does work via NVMe but not via USB. While with Ubuntu is the other way around. LibELEC is fine in both configs...

Poopi, 2024-01-27:
Adding log from USB for the reference 
[usb.tgz](https://github.com/procount/pinn/files/14072508/usb.tgz)
As suspected the root partition in not being mounted. Caused by the face it is an Extended Partition?

Poopi, 2024-01-27:
Adding logs from boot without PINN
[no_pinn.tgz](https://github.com/procount/pinn/files/14072696/no_pinn.tgz)

Poopi, 2024-01-28:
It seems that this is very fishy:
>[    1.142820] kvm [1]: VHE mode initialized successfully
[    1.143543] Initramfs unpacking failed: invalid magic at start of compressed archive
[    1.148491] Initialise system trusted keyrings

Poopi, 2024-01-28:
After multiple trials and error I've made Ubuntu working with PINN and directly from NVME ;) - in kinda hacking way but...
I was suspecting something fishy with Ubuntu kernel/initrd compilation. I turned out it is true. 
I downloaded nightly 24.04 image and checked the system-boot partition. Interestingly it contained the same kernel as for 23.10 but the initrd was significantly smaller.  So basically I've copied both files to NVMe and bang it boots :)

procount, 2024-01-28:
Well done!
I was beginning to suspect the initrd but I couldn't understand why it would work on a Pi5/NVMe without PINN.

Poopi, 2024-01-28:
Me neither. But it seems like for some reasons the initramfs was corrupted during kernel boot. I was trying to load it in different memory regions but nothing helps. After booting and upgrading the packages to latest 23.10 I also noticed significant decrease in initrd.img file size. 37MB for 1005 down to 20MB in 1009 ... If I will find some time I will dig into Ubuntu sources to check.

procount, 2024-01-29:
The initrd change mostly seems to be a reduction in the number of power device drivers .....🤷‍♂️ Nothing majorly significant. Nevertheless, I have copied the initrd.img from 24.04 into 23.10 to create a new ubuntu2310a version (in Testing) so I would be grateful if you could try this out.
Plus I have created an install for Ubuntu24.04 in case something else needs changing I have missed.
To get access to these versions, please unzip the attached file and copy os_list_v3.json file onto your PINN recovery partition, then reboot PINN. (Once the new OSes have been installed and tested on NVMe, the os_list_v3.json file can be deleted to prevent future confusion)

[os_list_v3.zip](https://github.com/procount/pinn/files/14084677/os_list_v3.zip)

Poopi, 2024-01-29:
24.04 works like a charm. Are you sure you updated the initrd.img? It still shows Oct 12 as data stamp and the kernel module version is from 6.5.0-1005 kernel instead of 1009. Anyway It would be good to copy both kernel and modules from 1009 version.  There is an interesting fix in the kernel added recently [Kernel](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/2042573)

procount, 2024-01-29:
Hmm, I thought I had. Oh well. I'll have to give it another go. At least we have a way forward for NVMe :)

androot2, 2024-02-02:
>>Poopi,
could you share this initrd.img to replace in23.10? I'm newbie in linux systems... In software too... :)
I have the same problem with my PINN instaled on nvme disk.

procount, 2024-02-02:
I have rebuilt 23.10 installation by replacing with some files from 24.04 (which still seems to have the 1005 version)

Please unzip the attached file and copy os_list_v3.json file onto your PINN recovery partition, then reboot PINN. (Once the new OSes have been installed and tested on NVMe, the os_list_v3.json file can be deleted to prevent future confusion)
[os_list_v3.zip](https://github.com/procount/pinn/files/14139880/os_list_v3.zip)

Please try installing Ubuntu2310b from the Testing category onto your NVMe drive.

androot2, 2024-02-02:
Hello. I just installed 23.10b and it's ok (problem temporary solved :)
Thank you!

procount, 2024-02-02:
Brill!
Thanks for testing.
I will repkace the original 23.10 with this b version.

mann1x, 2024-02-05:
@procount 

Is there a known issue with big SD cards?

Had no problems with a 64GB Sandisk Extreme Pro.
I use https://pinn.mjh.nz/ to create a template for recovery.cmdline.

Now I bought a Kingston Canvas Go Plus 256GB and seems the recovery.cmdline is just ignored.

Is there any way to troubleshoot this?
I don't see anything useful in /tmp/debug.

procount, 2024-02-05:
V3.8.9 beta is a bit different.
Rename it to cmdline.txt instead of recovery.cmdline

mann1x, 2024-02-05:
> V3.8.9 beta is a bit different. Rename it to cmdline.txt instead of recovery.cmdline

Ouch, I probably did it last time and then forgot about it... thanks.

The first boot is still a lottery, same with the 64GB.
Sometimes goes smoothly but very often doesn't.
After a few windows informing it's reading and parsing the cmdline goes to the main window with the list of images selected to install.
Before the list is ready you have on top the small progress window with "PINN is initializing" in the middle.
It just halts at some point and the system freezes completely.

Did it twice now that could read the cmdline.txt; first two complete freezes, 3rd only crashing the UI (jumped out to console), then with a couple of lucky reboots I got it finally installing.

Never had any other freezing issue after the first boot to install.
Always been able to go through it just retrying again.

Will try again to reproduce it with the 64GB, which went way smoother, now that I can free it up for testing.

procount, 2024-02-05:
See #770. Try not to touch the mouse until it has fully initialised, got an ip address and read all files, to see if that helps.

theofficialgman, 2024-02-13:
pinn-lite.zip https://sourceforge.net/projects/pinn/files/pinn-lite.zip/download is still missing the pi5 dtb so cannot be booted.
are there plans on when this will be added?

procount, 2024-02-13:
The Pi5 requires more than just a dtb file, which is why it is still in beta.
Please see the announcement at https://forums.raspberrypi.com/viewtopic.php?p=2173713#p2173713
It is usable if you can live with its couple of issues.

theofficialgman, 2024-02-14:
> The Pi5 requires more than just a dtb file, which is why it is still in beta. Please see the announcement at https://forums.raspberrypi.com/viewtopic.php?p=2173713#p2173713 It is usable if you can live with its couple of issues.

@procount consider correcting your initial post then https://forums.raspberrypi.com/viewtopic.php?f=63&t=142574 scrolling to the bottom of that post leads to the download link that I posted above which does not have pi5 support. yet it says at the top that pi5 support is available in the beta. I looked nearly everywhere (apparently not every page on that thread) for the beta (github, sourceforge) and could not find it.

procount, 2024-02-14:
Post updated.

Fredfars, 2024-02-21:
Hi, I have the same issue with my Crucial P3 Plus connected to a Pimoroni Base NVMe
And i found this about last kernel. 
https://github.com/raspberrypi/linux/issues/5873
Thanks for your job! I’m impatient to have dual boot (PiOS et Ubuntu) on my Pi5 :)

theofficialgman, 2024-02-21:
> Post updated.

@procount I don't see any edits on the initial post https://forums.raspberrypi.com/viewtopic.php?f=63&t=142574

> Last edited by [procount](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=74876) on Thu Dec 28, 2023 2:35 pm, edited 96 times in total.

procount, 2024-02-21:
Must have forgotten to press submit. Updated again.

Dent2024, 2024-02-21:
Hello, is it possible that the bcm2712-rpi-5-b.dtb is missing in the ZIP file? Or am I doing something wrong?

[31 more replies]